eyal0/OctoPrint-PrintTimeGenius #231: PrintTimeGenius % vs the % passed on via Mqtt

jwaes, 2021-02-05:
OctoPrint Version: 1.5.3
PrintTimeGenius Version (if you know): 2.2.7

I am not sure whether this a PrintTimeGenius issue or Mqtt issue.

I have also installed the plugins MQTT (0.8.7) and HomeAssistant discovery (3.2.1)
And the % passed on there is not the same as shown in Ocotoprint ?
Is PrintTimeGenius keeping its own % and replacing it in the GUI, but the original Ocotoprint estimate % still exists in the background somewhere ?

denkyem, 2021-02-10:
Using the octoprint API to return the progress we have the same issue.

eyal0, 2021-02-11:
@jwaes Yes, you are exactly correct.

eyal0, 2021-02-11:
The percentage that PrintTimeGenius shows is modified to be (time so far) divided by (time so far + time remaining).  That's easy math that other plugins could do.

eyal0, 2021-02-12:
This can't be fixed by OctoPrint-PrintTimeGenius.  It needs to be fixed by each plugin.  Or better yet, OctoPrint should fix it.  Perhaps file a bug with OctoPrint.  They could have an option to display percentage by time instead of by file size.

closed 2021-02-12 by eyal0