eyal0/OctoPrint-PrintTimeGenius #234: Can't find: marlin-calc.windows-x86.exe

ED-JE-70, 2021-02-16:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:
1.5.3

PrintTimeGenius Version (if you know):
2.2.7

What did you try:
Try automatic and manual analysis

What happenned:
No stars. only a hourglass behind the files

What did you expect to happen:
Analyzing the files and show the golden star

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

Running 32bit Windows10

2021-02-16 10:41:34,403 Not running built-in analysis.
2021-02-16 10:41:34,404 Disabled: "c:\octoprint\venv\scripts\python.exe" "C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_gcode_comments.py" "C:\Users\Gebruiker\AppData\Roaming\OctoPrint\uploads\Rahmen_h1.gcode"
2021-02-16 10:41:34,404 Running: "c:\octoprint\venv\scripts\python.exe" "C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Gebruiker\AppData\Roaming\OctoPrint\uploads\Rahmen_h1.gcode" "M92 X80.0 Y80.0 Z800.0 E90.0
M203 X500.0 Y500.0 Z5.0 E25.0
M201 X2000.0 Y2000.0 Z10.0 E74.0
M204 S300.0 P750.0 R1000.0 T300.0
M200 S0 D1.75
M900 K0.01
M205 B20000.0 E5.0 S0.0 T0.0 X10.0 Y10.0 Z0.3 J0.0"
2021-02-16 10:41:35,465 Failed to run '"c:\octoprint\venv\scripts\python.exe" "C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_progress.py" marlin-calc "C:\Users\Gebruiker\AppData\Roaming\OctoPrint\uploads\Rahmen_h1.gcode" "M92 X80.0 Y80.0 Z800.0 E90.0
M203 X500.0 Y500.0 Z5.0 E25.0
M201 X2000.0 Y2000.0 Z10.0 E74.0
M204 S300.0 P750.0 R1000.0 T300.0
M200 S0 D1.75
M900 K0.01
M205 B20000.0 E5.0 S0.0 T0.0 X10.0 Y10.0 Z0.3 J0.0"'
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint_PrintTimeGenius\__init__.py", line 339, in _do_analysis
    raise Exception(results_err)
Exception: Running: "C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers\marlin-calc.windows-x86.exe" "C:\Users\Gebruiker\AppData\Roaming\OctoPrint\uploads\Rahmen_h1.gcode" "M92 X80.0 Y80.0 Z800.0 E90.0

M203 X500.0 Y500.0 Z5.0 E25.0

M201 X2000.0 Y2000.0 Z10.0 E74.0

M204 S300.0 P750.0 R1000.0 T300.0

M200 S0 D1.75

M900 K0.01

M205 B20000.0 E5.0 S0.0 T0.0 X10.0 Y10.0 Z0.3 J0.0"

Can't find: C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers\marlin-calc.windows-x86.exe

2021-02-16 10:41:35,467 Disabled: "c:\octoprint\venv\scripts\python.exe" "C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers/analyze_gcode_comments.py" "C:\Users\Gebruiker\AppData\Roaming\OctoPrint\uploads\Rahmen_h1.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

eyal0, 2021-02-17:
I never thought to build it!  I guess that I could.  It needs to be added here: 

https://github.com/eyal0/Marlin/blob/estimate-bugfix-2.0.x/Marlin/SConstruct#L67

If I make it, can you test it?

ED-JE-70, 2021-02-18:
Yes of course!! Thank you....

eyal0, 2021-02-18:
I have something for you to test.  Please install PTG from zip file.  You can install it from the OctoPrint Plugin Manager like usual, but instead of selecting PrintTimeGenius from the repository, you will put this URL:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/win32_support.zip

You do **not** need to uninstall your existing PrintTimeGenius.  You will **not** need to uninstall the zip when the new release comes out, either.

Please check the log just like last time to see that it works.  Also please confirm that this file exists on your computer: C:\OctoPrint\venv\Lib\site-packages\octoprint_PrintTimeGenius\analyzers\marlin-calc.windows-x86.exe

After you confirm that it works, I'll release with this in place.

ED-JE-70, 2021-02-19:
Thanks!! files are analyzed and marked with a gold star. First print had an estimation of 37minutes but it actually took 30minutes to print......i will look into that but thanks for the update...!!!!!

eyal0, 2021-02-19:
PTG isn't able to estimate how long it will take your printer to heat up. It will learn this over time.

Make sure to run M503 one time so that PTG can see all your settings and use them in calculations.

PR #236 (2021-02-23): Add win32 binary
This fixes #234
closed 2021-02-23 by eyal0