eyal0/OctoPrint-PrintTimeGenius #237: Insert GCode for Progress on Display

rs-development, 2021-02-18:
Is it possible to replace progress GCode generated from the slicer? 

Goal is to get an ETA on the printer display that is the same as in the Plugin. Not from the slicer.

If not possible to replace the existing GCode is it possible to generate new GCode commands from the Plugin. Both the remaining time and the local time  when it will finish would be nice.

In Klipper you can define custom GCode Commands in the printer config. I don't know how this is with other firmwares.

eyal0, 2021-02-18:
Plugins that modify gcode are bad news.  I really dislike them because when there is more than one installed, they compete for modifying the gcode simultaneously and it causes problems.  Some of the bug reports that I get are due to this.

However, it is possible to add commands into the command stream without modifying the gcode.  I think that the DisplayLayerProgress plugin might work this way, though I'm not sure.

If DisplayLayerProgress is only showing the file progress (that is, bytes complete out of total file bytes), then it ought to have an option to report time printing so far out of total print time.  That would fix the problem without modifying gcode.

rs-development, 2021-02-20:
I am with you about modifying GCode!

I have found the display layer progress plugin and tested it, it shows the correct Percentage. So I will check if I can rewrite/configure that to send a Custom GCode with integers instead of text. It uses the message command at the moment, which overwrites over messages.

KevinNovak, 2021-05-19:
You can use [Display-ETA](https://plugins.octoprint.org/plugins/display_eta/) along with PTG to achieve this. It can display the ETA right on the printer display, using the time that PTG calculates.

eyal0, 2021-05-19:
I think that it should work, both easier and better than modifying gcode.  I'll close this issue.  Re-open it if you have a problem with the solution proposed.

closed 2021-05-19 by eyal0