eyal0/OctoPrint-PrintTimeGenius #238: [Question] Is PrintTimeGenius Arc or Circle Move aware?

LMS0815, 2021-02-20:
Seems the time calculation is wrong if the GCODE file was optimized with [ArcWelderPlugin](https://github.com/FormerLurker/ArcWelderPlugin).
Might be some timing or missing Marlin G2/G3 support.

eyal0, 2021-02-20:
I added G2/G3 support not long ago and I tested it thoroughly. I even noticed some inaccuracies in Marlin firmware and improved the arc code there.

Can you send an example of an arc that isn't being computed correctly? Best if you sent the file both before and after arc welder so that we can compare the difference in prediction.

LMS0815, 2021-02-21:
Unfortunately I deleted the file after printing. I was just wondering if your plugin was having an issue as the furures have been massively wrong.
The calculation was claiming only 60% ready at the end of the print job.

If I'll find again such a gap, I'll open a new issue. Thank you for your reply!

closed 2021-02-21 by LMS0815