eyal0/OctoPrint-PrintTimeGenius #239: Update OctoPrint for the print time?

kennethjiang, 2021-02-20:
OctoPrint Version: 1.5.3

Hi @eyal0, I have been assuming PrintTimeGenius plugin would update the times in OctoPrint. Maybe it's not the case?

I just opened a ticket for TSD plugin: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/107

But I wondering if PrintTimeGenius plugin can update the times in OctoPrint, so that it's available for TSD, as well for any other plugins?

If that's not an option, I'll be happy to make special integration with  PrintTimeGenius plugin so that the users who have both TSD and  PrintTimeGenius plugins installed will have consistent print time readings.

eyal0, 2021-02-20:
PTG is updating the times reported by OctoPrint. PTG fulfills the "estimator" contract in the OctoPrint hooks. So it replaces OctoPrint's estimation. All other plugin should see that estimation if they get the time left to print in the normal way.

Are you seeing that the time to print remaining is correct in the web UI and wrong in the plugin or is it wrong in both or...?

kennethjiang, 2021-02-21:
I haven't tested it myself. But a user reported he was seeing a much longer estimated print time in TSD (which gets the times from OctoPrint unless there is a bug) than that in OctoPrint. When he disabled PrintTimeGenius, these 2 times became consistent.

Let me do some testing myself to look at what's going on. Thank you so much for your prompt response @eyal0 !

eyal0, 2021-02-21:
No problem. You can turn on debug logging to see more of the data that PTG is reporting for run times.

Also, a lot of users of PTG get tripped up by the compensation. When PTG detects that its estimation is consistently wrong, it applies a factor to future estimations. That compensation can get very wrong if something weird happens, like a print that is paused or something. Compensation can be reset in the settings. The compensation also shows up in the PTG log, even without debug enabled.

Daskew78, 2021-02-27:
I'm not sure if this is related or a different issue but I just noticed that the iOS app with Octopod is showing a different percentage than the web interface does. All other stats appear to be correct apart from the percentage. I raised the issue with Gaston Dombiak who asked me to raise the issue with yourselves stating:

> Thanks David Askew for reporting this. OctoPod shows progress reported by OctoPrint. Are you using some plugin that is doing its own calculation for print progress? I think that there is a plugin (printtimegenius?) that does its own calculation and changes the UI but might not be changing OctoPrint itself so OctoPrint reports something different. IOW, not much I can do. If you can find the plugin that is changing the progress then maybe contact the developer of the plugin and ask to change OctoPrint progress in the server itself and not only the UI? I have not looked deep into this so my conclusion might be wrong 

Does this sound like it might be the same issue?

eyal0, 2021-02-27:
Everything that he writes is 100% correct. Only one other thing: PTG cannot modify OctoPrint's reporting of percentage. OctoPrint would need to do that.

I suggest that you file a bug at OctoPrint to do that.

Daskew78, 2021-02-27:
I'll pass that on, thanks!

eyal0, 2021-02-28:
No problem.  I get this question frequently.    I think that everyone is at first surprised to learn that the progress percentage reported by OctoPrint is bytes and not time.  It could be that OctoPrint chose this method because that was what was available first and they just stuck with it?  Or maybe the time-based progress was not accurate enough for reporting?

OctoPrint replaces that display with a time-based one.  Maybe I never should have added that feature because it causes so many questions like yours!

It's really not OctoPrint's fault.  OctoPrint reports file progress but it also reports the print time left and the print time so far.  Any plugin that wants to use time-based progress instead of file-based progress just needs to take the two numbers and divide.  It's trivial to implement.  If there's a plugin that isn't doing this but should, I'd take it up with them.

Whether OctoPrint would make the change is questionable.  Who know, maybe there are users that are expecting that progress meter to be file-based and not time-based and it will screw something up for them?  OctoPrint *could* report both and maybe that would spur developers to use it.  But again, OctoPrint already reports the time and time left so developers could have used that anyway.

Are the two progress numbers even that far apart?

kennethjiang, 2021-03-01:
Let me close this issue as PTG behaves as expected.

closed 2021-03-01 by kennethjiang