eyal0/OctoPrint-PrintTimeGenius #241: Can PTG get rid of the logging dependency altogether?

foosel, 2021-02-22:
This is not really an issue with PrintTimeGenius itself but rather with its Python 2.7 only dependency `logging`. Apparently that one has a namespace conflict with Python's own standard library `logging` module, causing issues deep within `pip` if installed in the same venv. We've seen some obscure update issues due to that on the forums and on Discord, one example [here](https://community.octoprint.org/t/problem-upgrading-to-release-1-5-3/30135). It would be lovely if this dependency could be avoided altogether in future versions, especially since its presence really makes no sense at all considering that PEP282 and thus a standard `logging` module has been implemented in every single Python version that OctoPrint even supports. Coincidentally it was last updated 8 years ago and its homepage is defunct.

---

OctoPrint Version: any

PrintTimeGenius Version (if you know): any

What did you try: Updating OctoPrint from 1.4.0 to 1.5.3

What happenned: It failed due to the `logging` dependency causing a name space conflict with Python's own standard library `logging`  module and thus causing failures deep within pip itself:

```
Traceback (most recent call last):
File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
"main", fname, loader, pkg_name)
File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
exec code in run_globals
File "/home/pi/oprint/lib/python2.7/site-packages/pip/main.py", line 16, in
from pip._internal.main import main as _main # isort:skip # noqa
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/main.py", line 13, in
from pip._internal.cli.autocompletion import autocomplete
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/autocompletion.py", line 11, in
from pip._internal.cli.main_parser import create_main_parser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/main_parser.py", line 7, in
from pip._internal.cli import cmdoptions
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/cli/cmdoptions.py", line 25, in
from pip._internal.locations import USER_CACHE_DIR, get_src_prefix
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/locations.py", line 18, in
from pip._internal.utils import appdirs
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/appdirs.py", line 16, in
from pip._internal.utils.compat import WINDOWS, expanduser
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_internal/utils/compat.py", line 17, in
from pip._vendor.urllib3.util import IS_PYOPENSSL
File "/home/pi/oprint/lib/python2.7/site-packages/pip/_vendor/urllib3/init.py", line 21, in
from logging import NullHandler
ImportError: cannot import name NullHandler
```

What did you expect to happen:

No third party dependencies from a third party plugin nuking standard Python libraries, causing update failures in the progress.

eyal0, 2021-02-22:
I have this in setup.py:

```
if version_info[0] < 3:
    plugin_requires.append("logging")
```

Is that not enough?

eyal0, 2021-02-22:
I'm not an expert on python version stuff.  Just tell me what to do and I'll do it.  :-)

foosel, 2021-02-22:
That means it's still getting installed on Python 2. You don't need it at all really because all that this package does is stuff that is already contained in every Python version that OctoPrint itself supports, as part of the standard library, so you can just get rid of it altogether and remove the risk of it causing nasty side effects.

cp2004, 2021-02-22:
This is a massive mystery that has bugged me for a while... and is now solved. I always just assumed someone messed up their installation somehow. TBH I don't think PyPi should allow for packages that take the place of standard library, call it something else (see urllib vs urllib2 and urllib3, for example). But anyway, getting rid of that package would be a step in the right direction.

PR #242 (2021-02-23): Get rid of unnecessary logging dependency
`logging` has been part of Python's standard library since forever (and certainly for all versions of Python that OctoPrint will run under). There's no need to pull in a third party implementation of it that clashes with the standard library version and actually causes issues within `pip` itself.

Closes #241
commit d9f36ea1 (2021-02-23): Get rid of unnecessary logging dependency

`logging` has been part of Python's standard library since forever (and certainly for all versions of Python that OctoPrint will run under). There's no need to pull in a third party implementation of it that clashes with the standard library version and actually causes issues within `pip` itself.

Closes #241
closed 2021-02-23 by eyal0