eyal0/OctoPrint-PrintTimeGenius #243: Filament (Tool 0) Length Incorrect

ZhuDaHai, 2021-03-04:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: OctoPrint 1.5.3, Python 2.7.16, OctoPi 0.17.0

PrintTimeGenius Version (if you know): 2.2.7

What did you try:
I load a sample 3DBenchy.gcode file and OctoPrint reports the incorrect Filament (Tool 0) Length of 1.98m
The correct length, as reported by S3D and recorded in the gCode files is: 4.78m

What happenned:
The incorrect length was reported.
After disabling PTG, the correct length was reported in OctoPrint

What did you expect to happen:
I expected the correct filament length to be reported. I have 2 other OctoPrint servers on 2 other 3D Printers. They PTG version and OctoPrint, Python, OctoPi versions are identical to this one, yet they report the correct filament length even with PTG plugin enabled. 

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
With PTG enabled:
[image]

With PTG disabled:
[image]

S3D Filament Length Estimate:
[image]

Gcode file entry:
[image]

All the PTG settings across all three of my printers are identical, yet this one PTG instance reports the wrong filament length. Therefore I can only assume it is an interaction with another plugin or system setting (that I cannot find: I have looked and looked) that is causing this issue. 

I have also verified that the M200 command, as listed under M503, displays the proper filament size (1.75) as set in the firmware (as it is with all 3 3D Printers). 

Hopefully you can point me to where I've missed a setting, or fix this little issue.

Thank you.

eyal0, 2021-03-04:
Please send the PTG logs. Sounds like PTG doesn't have the right M200 info like you suspected but let's make sure.

ZhuDaHai, 2021-03-05:
Attached is the log. I did a quick search and found M200 D1.75 throughout. So I'm not sure what's going wong:
[plugin_PrintTimeGenius_engine.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6087619/plugin_PrintTimeGenius_engine.zip)

eyal0, 2021-03-05:
M200 D1.75 means that you're in volumetric mode.  Is that what you expected?

https://marlinfw.org/docs/gcode/M200.html

Most people are *not* using volumetric mode and it says `M200` in their `M503` twice, once to specify diameter and then once to disable the volumetric mode.  This can cause the error that you're seeing.  Notice how the number for length (in meters) is swapped with the volume (in cm cubes).

Could you please also send the output of `M503`?  Sometimes the Marlin folks change how M200 works and maybe my code isn't up-to-date on that.

ZhuDaHai, 2021-03-05:
All the Marlin firmware in my printers are set up this way:

```
// Generally expected filament diameter (1.75, 2.85, 3.0, ...). Used for Volumetric, Filament Width Sensor, etc.
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
```
```
/**
 * Disable all Volumetric extrusion options
 */
//#define NO_VOLUMETRICS
```
Yet only the A6 cannot accurately calculate the Filament (Tool 0) Length when PTG is enabled. And, yes, I understand that NO_VOLUMENTRICS has not been defines (I hate these Negative defines, BTW). But read on...

I compared the same log file section between my A6 (wrong length calculation) with the A4 (correct length calculation) and they both appear to be identical:

I compared the PTG log files between the A6 (wrong length) and the A4 (correct length) and found this:

**A6 (wrong filament length):**
```
2021-03-05 09:32:33,708 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/3DBenchy_-_Skin_TPU_A6.gcode" "M200 D1.75
```

**A4 (Correct Filament Length):**
```
2021-03-04 07:12:58,618 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/3DBenchy_-_Skin_TPU_A6.gcode" "M200 S0 D1.75
```

Notice the A6 does not have the 'S0' parameter for M200, while the A4 does.

BUT..wait! Its gets stranger, the M503 command on both printers report the same M200 S0 D1.76 , and also states that Filament settings are Disabled:

**A6 (wrong length):**
```
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75

```

**A4 (correct length):**
```
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
```

So where and HOW is PTG reading an M200 without the S0 parameter?

here's the complete M503 for the A6 (wrong):
```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X160.00 Y160.00 Z1600.00 E196.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X1200.00 Y1200.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1200.00 R1000.00 T1200.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.01
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Mesh Bed Leveling:
Recv: echo:  M420 S0 Z2.00
Recv: echo:  G29 S3 I0 J0 Z-0.65000
Recv: echo:  G29 S3 I1 J0 Z-0.36000
Recv: echo:  G29 S3 I2 J0 Z-0.22000
Recv: echo:  G29 S3 I3 J0 Z-0.22000
Recv: echo:  G29 S3 I4 J0 Z-0.35000
Recv: echo:  G29 S3 I5 J0 Z-0.65000
Recv: echo:  G29 S3 I0 J1 Z-0.52000
Recv: echo:  G29 S3 I1 J1 Z-0.23000
Recv: echo:  G29 S3 I2 J1 Z-0.09000
Recv: echo:  G29 S3 I3 J1 Z-0.09000
Recv: echo:  G29 S3 I4 J1 Z-0.23000
Recv: echo:  G29 S3 I5 J1 Z-0.50000
Recv: echo:  G29 S3 I0 J2 Z-0.52000
Recv: echo:  G29 S3 I1 J2 Z-0.23000
Recv: echo:  G29 S3 I2 J2 Z-0.08000
Recv: echo:  G29 S3 I3 J2 Z-0.09000
Recv: echo:  G29 S3 I4 J2 Z-0.25000
Recv: echo:  G29 S3 I5 J2 Z-0.55000
Recv: echo:  G29 S3 I0 J3 Z-0.75000
Recv: echo:  G29 S3 I1 J3 Z-0.40000
Recv: echo:  G29 S3 I2 J3 Z-0.25000
Recv: echo:  G29 S3 I3 J3 Z-0.25000
Recv: echo:  G29 S3 I4 J3 Z-0.25000
Recv: echo:  G29 S3 I5 J3 Z-0.75000
Recv: echo:  G29 S4 Z0.00000
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H200 B60 F0
Recv: echo:  M145 S1 H210 B0 F0
Recv: echo:; PID settings:
Recv: echo:  M301 P23.94 I1.52 D94.27
Recv: ; Controller Fan
Recv: echo:  M710 S255 I128 A1 D60 ; (100% 50%)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X800 Y800 Z800
Recv: echo:  M906 T0 E900
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U100.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S1
Recv: ok
```

And the A4 (correct length):
```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X160.00 Y133.30 Z800.00 E196.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E100.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P750.00 R3000.00 T1000.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.01
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210 F0
Recv: echo:  M145 S1 H240 F0
Recv: echo:; PID settings:
Recv: echo:  M301 P21.18 I1.18 D94.83
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X800 Y800 Z800
Recv: echo:  M906 T0 E950
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U100.00
Recv: ok
```

Lastly, the Marlin source versions are:
A6 (wrong length): 020006
A4 (correct length: 020007

Hope this helps. Thanks for your time!

ZhuDaHai, 2021-03-05:
Ok, here's some more strangeness: I sliced a Grogu head (75% size) and S3D says its 10042mm in length - and PTG agrees! (On the A6)! Yet the 3DBenchy is still wrong (at half of the length that it should be). 

S3D:
[image]

PTG:
[image]

And here are those parts of the log files from PTG.
Grogu head:
`2021-03-05 13:34:35,547 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/head_-_Green_TPU_A6.gcode" "M200 S0 D1.75`

Same PTG Log, 3DBenchy:
`2021-03-05 09:32:33,708 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/3DBenchy_-_Skin_TPU_A6.gcode" "M200 D1.75`

Notice how the 'S0' is present on the Grogo head, but absent on the 3DBenchy. All I've done it reboot the A6 3D Printer - but I did that multiple times yesterday and today with the 3DBenchy with still the same wrong length results.

Attached is the entire log file:
[plugin_PrintTimeGenius_engine (2).zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6088363/plugin_PrintTimeGenius_engine.2.zip)

ZhuDaHai, 2021-03-05:
P.S. The A6 (with the flaky Length reporting) is running Marlin 2.0.6
The A4 (with correct Length reporting) is running Marlin 2.0.7.2

eyal0, 2021-03-05:
Good job with the debugging.

Looks like PTG isn't seeing the `S0` in there and that's the problem.  Not sure why it wouldn't see it, though.

Maybe it has to do with rebooting?  PTG constantly monitors the console looking for that `M200` command so that it can remember what settings to use for measuring the time to print with marlin-calc.  However, to keep from trashing the disk, it won't save it into the config until the OctoPrint is shutdown.  Maybe the saving isn't working?

The config is stored in a file called "printer_config.yaml", maybe in the directory called `~/.octoprint/data/PrintTimeGenius`.  Can you see if the file is there?  Maybe PrintTimeGenius is having trouble re-writing the file so each time you reboot it reverts to old settings until you run M503 again and it gets the new ones?

This is also a quick solution for you: If OctoPrint won't overwrite that file properly then just modify it yourself.

Let me know what you discover, thanks!

ZhuDaHai, 2021-03-06:
Ok, this are getting weirder. 
I found the PTG yaml file (below):
```
printer_config:
- M200 S0 D1.75
- M92 X160.00 Y160.00 Z1600.00 E196.00
- M203 X300.00 Y300.00 Z5.00 E25.00
- M201 X1200.00 Y1200.00 Z100.00 E1000.00
- M204 P1200.00 R1000.00 T1200.00
- M205 B20000.00 S0.00 T0.00 J0.01
version: 2.2.7
```
It all looks good.
So I renamed it to `printer_config_yaml.old`
After many reboots, restarts, power off/on, it never created a new file. However, the Filament (Tool 0) Length is still showing correct for the 3DBenchy every time I reload it. 
And now I'm seeing these errors, occasionally, after rebooting/restarting:
[image]
[image]

It only wrote a new printer_config.yaml after I did an M503 in the Termal tab:
```
printer_config:
- M200 S0 D1.75
- M92 X160.00 Y160.00 Z1600.00 E196.00
- M203 X300.00 Y300.00 Z5.00 E25.00
- M201 X1200.00 Y1200.00 Z100.00 E1000.00
- M204 P1200.00 R1000.00 T1200.00
- M205 B20000.00 S0.00 T0.00 J0.01
version: 2.2.7
```
So this is kinda odd: shouldn't it automatically build the yaml file if its missing? especially after a reboot? or new gcode file load?

Perhaps you nailed it on 'saving isn't working' as my SD Card has become corrupted. I'm very careful about shutting down OctoPrint before powering off, but have not been perfect with this - especially when something jams and I have to quickly power off.

Also, this is a 64GB SD Card. Have you heard of any issues using one that large?

I will reformat this OctoPrint card and reload all the settings from a backup (hopefully the backup is not corrupt) and do more testing.

ZhuDaHai, 2021-03-07:
Even after loading up a new 64GB SD Card with OctoPi and restoring my backup, I'm still getting those funky errors about renders and update checking. But I've not had any further issues with Filament (tool 0) lengths being incorrect. 
But now my wifi signal quality has gone to crap. Its not nearly as fast as before and the webcam stream takes forever to load.
Gremlins
Gotta be Gremlins
So, you can close this as a non-issue, since I can no longer reproduce it. And now I have other issues not present before. 
Ugh.
Thanks again for your time, patience and input!

eyal0, 2021-03-07:
3D printing is never just a set-it-and-forget-it device. Neither are laser printers, though, do I guess that we can expect decades of work ahead of us.

Enjoy!

closed 2021-03-07 by eyal0