eyal0/OctoPrint-PrintTimeGenius #244: Very wrong print time left.

thetof91, 2021-03-07:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.5.3

PrintTimeGenius Version (if you know): 2.2.7

What did you try: 

What happenned: Print time left is very wrong. In cura it showed would take about 8 hours, After 43 min of print, print time left is 18 hours and 36 mins. And the time go down fast at some points. Have notice also with smaller prints it is very wrong in the other way, it shows 1 min left, and then all of the sudden adds 5 mins, and when those 5 mins done add 5 mins more.

What did you expect to happen: Show more correct time to print.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2021-03-07:
I'll probably need log files to diagnose this. Often the problem is that the M503 output isn't in use or there is some bad compensation values in the settings.

ilium007, 2021-03-08:
Does M503 have to be run on every print?

eyal0, 2021-03-08:
No, just once.  PrintTimeGenius will notice it and store the result.  Some printers do it when they boot up so you might already have it.