eyal0/OctoPrint-PrintTimeGenius #245: Analysis from SD card may fail

bitbckt, 2021-03-08:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.5.3

PrintTimeGenius Version (if you know):  2.2.7

What did you try:

Installed PTG on a new OctoPrint installation. Chose a file from my printer's SD card in the OctoPrint Settings -> PrintTimeGenius Plugin -> Analyze dialog. Clicked "Analyze".

What happenned:

No analysis occurred. The following error is in octoprint.log:

2021-03-08 13:11:29,589 - octoprint - ERROR - Exception on /plugin/PrintTimeGenius/analyze/sdcard/mflink~1.gco [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 520, in analyze_file
    queue_entry = self._file_manager._analysis_queue_entry(origin, path)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 1058, in _analysis_queue_entry
    absolute_path = self._storage(destination).path_on_disk(path)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 1038, in _storage
    "No storage configured for destination {destination}".format(**locals())
octoprint.filemanager.NoSuchStorage: No storage configured for destination sdcard

What did you expect to happen:

The file is analyzed.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2021-03-08:
Some sort of the problem in the OctoPrint filemanager?  I'd reboot and try again.

Is there anything peculiar about your setup?  Like the disk is read-only or something like that?

bitbckt, 2021-03-08:
I've tried rebooting both the Pi and the printer - no luck. The card is unlocked, and I've physically removed and reinserted it as well.

How might I check if Octoprint can write to the card, in general? It's not mounted as a volume on the Pi - it's on the printer.

bitbckt, 2021-03-08:
I uploaded a sample model directly from the UI, which seemed to work fine:

```2021-03-08 14:06:39,462 - octoprint.filemanager.analysis - INFO - Starting analysis of local:nefertiti_015mm_pla_mk3_7h52m.gcode
2021-03-08 14:06:39,701 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-08 14:06:39,959 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Sending file to SD"
```

The UI showed progress, as the file was transferred to the SD.

bitbckt, 2021-03-08:
```
octoprint.filemanager.analysis - INFO - Analysis of entry local:nefertiti_015mm_pla_mk3_7h52m.gcode finished, needed 24.74s
```

Analysis seems to work fine from local storage on the Pi. I can live with that, but checking whether the plugin worked using the dialog in PrintTimeGenius Plugin led me to try to analyze a file I had already loaded (on the printer's SD card). Should that work, or should I not expect that to work?

eyal0, 2021-03-08:
I think that it might be impossible.  This is a limitation of OctoPrint, not of PrintTimeGenius.  The reason is that the result of the analysis needs to be stored somewhere that OctoPrint can control, which is usually the `~/.octoprint` directory.  If your file isn't in that directory, OctoPrint doesn't have a standard place to store the metadata about that file.

I think that's the reason.  There's also other stuff that breaks when you try to print from SD card.  For example, OctoPrint doesn't even know the file size so it probably won't show you the progress.  I haven't tried it in a while.

bitbckt, 2021-03-08:
Interesting. Thanks for the insight and the fast response. I just stumbled into a strange corner condition on this.

Closing this out. Cheers.

closed 2021-03-08 by bitbckt