eyal0/OctoPrint-PrintTimeGenius #249: Error when fetching print history in `self.onBeforeBinding`

Nesciosquid, 2021-04-11:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

**Describe the bug**
Looks like PTG is hitting a null reference error when fetching print history on my Octoprint instance. (I've used PTG successfully on other instances, but this is a fresh install!)

It's failing to analyze (either automatically on uploads or when selecting files manually in the plugin settings), and there's almost nothing in the log, presumably because logging is being interrupted by the NPE.

[image]

Running the same command from the Chrome console causes the same error, and printing the value passed to `Octoprint.get().done()` shows that the response is `null`.

[image]

```
**To Reproduce**
Steps to reproduce the behavior:
1. Refresh page with PTG installed -- the error happens immediately on Octoprint page load. 

**Expected behavior**
PTG doesn't throw an error, and can analyze files. Secondarily, PTG should write to its log when it fails to fetch print history.

**OctoPrint Version:** OctoPrint: 1.5.3 (stable)

**PrintTimeGenius Version (if you know):** 2.2.7 (according to Plugin Manager)

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename the files).  **This is really important.  I probably can't help you without this.**

[octoprint (3).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6291093/octoprint.3.log)
[plugin_PrintTimeGenius_engine (4).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6291094/plugin_PrintTimeGenius_engine.4.log)
![PTG_options](https://user-images.githubusercontent.com/4042668/114290802-255f5d00-9a50-11eb-819d-52908231f461.PNG)

Nesciosquid, 2021-04-11:
Seems like maybe PTG is assuming that the print history has already been initialized -- maybe it behaves weirdly if you install it while a print job is actually running and it's failing to do its initial setup, or something like that?

eyal0, 2021-04-11:
That sounds plausible, yeah.  How to fix it... Hm...

Nesciosquid, 2021-04-11:
Print just finished, so while the printer was idle, I tried:

* Uninstall PTG (with the "clean" button)
* Restart Octoprint
* Install PTG
* Restart Octoprint

Still no luck -- error immediately shows up again.

Is there a way I can manually populate this files list, or try a previous version?

Nesciosquid, 2021-04-14:
Still can't get this to work. Suggestions?

I could re-install Octoprint and try installing PTG as the first (and only) plugin, maybe?

eyal0, 2021-04-14:
Want to try installing this and see if it is better?

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/refs/heads/print_history_error.zip

Nesciosquid, 2021-04-18:
Nope, still getting the same error. :(