eyal0/OctoPrint-PrintTimeGenius #250: Estimates immediately wrong

Jack77777777, 2021-04-12:
Estimated time was over 6h. after printing in 3:38h estimate shows now 5+h but still way to wrong.
[image]
[image]
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6298024/plugin_PrintTimeGenius_engine.log)
[Makita_Battery_New.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6298030/Makita_Battery_New.zip)

eyal0, 2021-04-12:
It seems that you had some recent prints that were much faster than predicted so PTG has adjusted.

Can you look in the advanced settings to see them?  If you delete them and then re-analyze the file, you will get a better result.

```
2021-04-12 11:47:30,559 Time spent printing, actual vs predicted: 12704.528058577998/12724.484557649725, 7098.386946807/18643.968241987117, 1569.4977473090003/4676.17500838264, 367.7750795760003/1300.4533922820253, 4728.910934346997/12855.435260625782
2021-04-12 11:47:30,565 Average scaling factor: 0.5272674592345831
```

This is in the file that you sent me.  See how it has scaled the time by 0.52?  That's is 52%, so the estimate will be 52% of actual.

eyal0, 2021-04-12:
I just added more info here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Jack77777777, 2021-04-12:
Does PTG notice a cancelled print or does it mean was finished faster then expected?

eyal0, 2021-04-12:
I don't even remember.  I think that it only looks at completed prints.  If you notice a different behavior, I can try to fix it.

(I developed this plugin back when I owned a 3d printer.  I haven't owned one for 3 years now so testing is made difficult.)

Jack77777777, 2021-04-13:
Deleted compensation values and reanalyzes all gcodes. Looks good so far.

eyal0, 2021-04-13:
Any guess what might have cause PTG to think that the print took less time than predicted?  Did you cancel a print mid-way?

Jack77777777, 2021-04-13:
I cancelled a job several minites after start due to Adhesion problems 🙈

eyal0, 2021-04-13:
Hm, that is surprising to me.  In the code, I am only recording into the print_history when the event is `PrintDone`:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/__init__.py#L474

And according to the documentation, `PrintDone` should only happen when a print is completed successfully.

https://docs.octoprint.org/en/master/events/index.html#printing

Yet you saw print history being recorded even when the print was cancelled.  Are you sure that the print history was written even when you cancelled the print?  If you can confirm this then I can see what the problem is.  Maybe the issue is in OctoPrint and not in PTG?

Jack77777777, 2021-04-14:
> Are you sure that the print history was written even when you cancelled the print?  If you can confirm this then I can see what the problem is.  Maybe the issue is in OctoPrint and not in PTG?

Where can i check this?

eyal0, 2021-04-16:
Start a print, wait a few minutes, cancel it in the middle, and let me know if it put a row in the print history table or not.

Then do it again with a print that you *don't* cancel in the middle and see if it puts a row in the print history table.

The table is viewable in the settings, like before.

That would prove one way or another if cancelled prints are getting added to the print history table.  I think that it would work!

Jack77777777, 2021-04-20:
Do you mean if it adds a row to https://plugins.octoprint.org/plugins/PrintJobHistory/ ?
I have enabled to save only successful jobs in https://plugins.octoprint.org/plugins/PrintJobHistory/

eyal0, 2021-04-20:
No, I mean the print history that PrintTimeGenius maintains inside of the PrintTimeGenius Advanced Settings.  It's only supposed to save on success but I don't know if it does that correctly and I'm curious to know.  In my testing it works fine but maybe you saw a different behavior that led to this bug report?

Jack77777777, 2021-04-23:
I did a snapshot of compansation values, cancelled a print and did another snapshot. Both have the same values.

Before Cancel:
[image]

After Cancel:
[image]

eyal0, 2021-04-23:
So it looks like everything is working okay now.  If you see more issues, go ahead and file a bug.

It's odd to me the words on your screen.  Does "Geasmt" mean "total" in English?  I guess that it translated only one word!  Surprising to me.

I'm also excited to see how good the predictions are.  Extruding prediction is very high quality.  I'm glad that my plugin is working.

Jack77777777, 2021-04-23:
Yes, Gesamt is german and means total. I think you can close this one. 

Thanks a lot for your effort and support.

closed 2021-04-26 by eyal0