eyal0/OctoPrint-PrintTimeGenius #252: Print progress wrong in 3rd party app

crysxd, 2021-04-18:
**Describe the bug**
I'm the developer of [OctoApp](https://play.google.com/store/apps/details?id=de.crysxd.octoapp&hl=nl&gl=US). One of my users reached out to report that the app is showing a different progress than the web interface. After some investigation it turned out that she had the PrintTimeGenius plugin installed. The app is relying on the [push updates from the rest API](https://docs.octoprint.org/en/master/api/push.html#id4). The progress reflect in the data there is still the "original" progress, not the updated one from your plugin. The updated time/eta estimations reflect as expected in rest API and the app.

Is there any way for you to also reflect the updated progress in the rest API? If not, is there any way the app can get the correct estimation?

**Expected behavior**
Data shown in webinterface and data provided in REST API is consistent

**OctoPrint Version:** 1.5.2/1.6-rc

**PrintTimeGenius Version (if you know):** 2.2.7

crysxd, 2021-04-18:
I found the issue and was able to fix it on my end. PrintTimeGenius changes the progress information shown to the user in the web interface through JS, but this obviously does not affect any 3rd party app or similar which consumes the API information. 

I now added a small piece of logic to my deserialization, if the `printTimeLeftOrigin` is `genius`, I overwrite the `completion` with the same formula used in the webinterface. While not perfect, this works for me. ([Source](https://gitlab.com/crysxd/octoapp/-/blob/d406f3c441a371fd13876bf8634850f2c56cde14/octoprint/src/main/java/de/crysxd/octoapp/octoprint/json/ProgressInformationDeserializer.kt))

You can close this issue at will, but I think it would be good to find a better solution as also other software is affected by this.

eyal0, 2021-04-18:
Everything that you write is correct. OctoPrint does not provide a time progress bar, only a bytes progress bar.

If you want to have the time progress then you can just do a simple division, remaining divided by total.

I think that OctoPrint doesn't provide this probably because the OctoPrint estimate isn't all that accurate in the first place.

I considered patching OctoPrint with this plugin but that kind of behavior from plugins is frowned upon.

I'll add a note into the wiki to describe this because you're not the first to notice it.

closed 2021-04-18 by eyal0

eyal0, 2021-04-19:
Updated the wiki to explain this.