eyal0/OctoPrint-PrintTimeGenius #253: Model height wrong

Jack77777777, 2021-05-11:
**Describe the bug**
If PrintTimeGenius is activated the height is wrong in web Ui and .metadata.json. If i deactivate PrintTimeGenius .metadata.json and Web Ui has the right data.

**To Reproduce**
Steps to reproduce the behavior:
1. Delete .octoprint/uploads/.metadata.json 
2. Activate PrintTimeGenius
3. Upload Gcode
4. Result:
[image]
[image]

5. Delete .octoprint/uploads/.metadata.json and Gcode
6. Deactivate PrintTimeGenius 
7. Upload same Gcode
8. Result:
[image]
[image]

**Expected behavior**
8.

**OctoPrint Version:**
1.6.1

**PrintTimeGenius Version (if you know):**
2.2.7

[Nut.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6457911/Nut.gcode.txt)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6457916/plugin_PrintTimeGenius_engine.log)

eyal0, 2021-05-11:
I see that PTG reports the "height" as "maxZ" minus "minZ", which is indeed 9.8mm.  Is it better to report 10mm?  I suppose so because that min height of 0.2mm is just the thickness of the layer.  I guess that for height I could just always report maxZ?

I'm also not accounting for nozzle thickness, I guess technically we should also be reporting X and Y to be enlarged by the thickness of the nozzle.  Ah well, no big deal.

eyal0, 2021-05-11:
Here's where things went wrong: https://github.com/OctoPrint/OctoPrint/commit/4c4570e2e168acedd91c67a68c1ba5e7748d5830

Maybe the heights reported by PTG and OctoPrint's built-in analyzer used to match before this change.  I can make the same change.

Jack77777777, 2021-05-11:
Yes, that's it. This is the reason why i referenced this issue there. I opened both issues. As OP is fixed now you have to match your code 😁.

eyal0, 2021-05-11:
If you'd like to look at the PR, be my guest.  If you want to download it, install this at a plugin:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/refs/heads/bedZ.zip

You will need to re-analyze files for it to take effect.

Jack77777777, 2021-05-11:
I will test tomorrow due to a ongoing print.

Jack77777777, 2021-05-12:
Tested and confirm it is working.

Thanks a lot

PR #254 (2021-05-12): Add bedZ feature, default 0
This fixes #253

This is related to https://github.com/OctoPrint/OctoPrint/commit/4c4570e2e168acedd91c67a68c1ba5e7748d5830
commit 3a33a463 (2021-05-12): Add bedZ feature, default 0

This fixes #253

This is related to https://github.com/OctoPrint/OctoPrint/commit/4c4570e2e168acedd91c67a68c1ba5e7748d5830
closed 2021-05-12 by eyal0