eyal0/OctoPrint-PrintTimeGenius #255: Feature Request - Analyze Button

the0remora, 2021-05-20:
I use SSH to upload my gcode this means that PrintTimeGenius does not automatically calculate the print time. Every time I have to go in to settings => Plugins => PrintTimeGenius find the gcode file in that list and hit analyze. Could an Analyze Button be added to the File Browser in the main window please?

eyal0, 2021-05-20:
Are you uploading into the correct folder? You should be uploading into the watched folder and not into the uploads folder.

the0remora, 2021-05-20:
I did not know that folder existed. Thank you.

closed 2021-05-20 by the0remora

eyal0, 2021-05-20:
Most people don't know about that folder!  I wish that more people did.  Maybe there ought to be a `README` file in the uploads folder explaining it.