eyal0/OctoPrint-PrintTimeGenius #256: [Request] Exclude files on SD card and/or make gcode filelist bigger

Gathaeryx, 2021-05-25:
The filelist in the plugin window is extremely small, and since there are only 5 lines to display, there is a lot of clutter from the files on my SD card, which can't be analyzed. It would be nice to either be able to make this window bigger, or to be able to exclude the SD card files to show up.

[image]

eyal0, 2021-05-30:
Hmm, yeah, that could be a good idea.  Because the files that are on the sdcard can't be analyzed anyway.  Okay, that should be easy, I think.

eyal0, 2021-05-30:
I haven't any way to test it.  Can you test it for me?  Please install this from the plugin settings within OctoPrint:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/refs/heads/exclude_sd_card.zip

And then check the list with the same SD card as before and see if it looks right to you.

Thanks!

Gathaeryx, 2021-08-18:
Sorry for the (extremely) late reply, but I just checked the new plugin, and the files on the SD card are indeed missing. Thanks! This makes it a lot easier to see which files still need analysing.

eyal0, 2021-08-18:
Fixed by #257

closed 2021-08-18 by eyal0