eyal0/OctoPrint-PrintTimeGenius #263: Error on analysis while printer is heating

subnoize417, 2021-08-10:
When attempting to analyze uploaded files during heatup, I get a python TypeError in the logs, shown below. Assuming it is due to the printer's current temperature being reported with a decimal.

Issue is present on both version 1.6.1 and the latest 1.7.0rc1 of Octoprint, running version PrintTimeGenius version 2.2.8.
Hardware is a Pi 3b+, running an up-to-date octopi 0.18.0 installation, and python 3.7.3.

```
Failed to run '"/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/REMOVED.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 329, in _do_analysis
    if self._aborted and not _allow_analysis(self._plugin._printer, self._plugin._settings):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 192, in _allow_analysis
    if thermostat['actual'] < thermostat['target'] - 5:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
```

I was able to resolve the issue myself by changing line 192 from `if thermostat['actual'] < thermostat['target'] - 5:` to `if int(thermostat['actual']) < thermostat['target'] - 5:`  to convert the float to an integer. Not familiar enough with git to create a pull request unfortunately, and the change has only been tested with one file so far

eyal0, 2021-08-10:
Duplicate of #210?

subnoize417, 2021-08-10:
Very similar error, yes. It appears it may have something to do with the PSU Control plugin, it hasn't been calculating the print time for me since I set that up. Steps I took to recreate the error: 
Removed & reinstalled PrintTimeGenius
Used 'Send to octoprint' in superslicer (fork of prusaslicer) with 'start printing after upload' checked. PSU control then turns on power to the printer, waits 15 seconds to connect, then beings the print automatically.

Did some more testing, attached the logs. With the file `printeronandconnectedtest.gcode` the plugin worked correctly, while the printer was powered on and connected. 2nd test, with the printer initially powered off, and powered on via PSU Control, and the file `printerofftest.gcode` I see the TypeError. Not sure what I did exactly by adding `int(` to line 192, I barely know python, just enough to "fix" an issue the completely wrong way

If I left something out of the logs below, let me know and I'll try and get you what you need.

[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6964422/octoprint.log)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/6964423/plugin_PrintTimeGenius_engine.log)