eyal0/OctoPrint-PrintTimeGenius #264: Wrong ETA-time in browser tab

ArnieO, 2021-08-20:
The ETA estimate in the browser tab is exactly one hour wrong (e.g. shows 15:10 in stead of 16:10).

**Expected behavior**
I expect the computer clock to be used as reference, adding "Print time left" to current computer time.

**OctoPrint Version:**
Version 1.6.1

**PrintTimeGenius Version:**
2.2.8

**Operating system etc:**
Windows 10 (fully updated)
The same error in several browsers: I tried Firefox, Chrome and Edge.

eyal0, 2021-08-20:
Is it always off by one hour?

ArnieO, 2021-08-20:
I just recently installed it. Noticed the issue yesterday, and it is the same today. 
Could it be a Daylight Saving Time thing?

eyal0, 2021-08-20:
Perhaps?

Usually OctoPrint doesn't show the time that the print will end, it shows how long the print will take. Which one are you seeing?

ArnieO, 2021-08-20:
The browser tab shows a clock time and a percentage finished. It is not showing the duration and not the time left. The time there fits with the finish ETA one hour off. Not printing now, so this is how it looks right now:
[image]

eyal0, 2021-08-24:
I don't know if that's a feature of OctoPrint or some other plugin that you've installed that is doing that.  ETA is not a feature of PrintTimeGenius and ETA is not a feature of OctoPrint, as far as I know.  It's been a while since I checked, though.

Maybe you can provide a list of other plugins that you have which might be doing it?

Also, off by an hour sounds like a timezone issue.  I would check the clocks and timezones in your browser and on your Raspberry Pi.  Do other websites cause you the same issue? https://time.is/

closed 2021-08-25 by ArnieO