eyal0/OctoPrint-PrintTimeGenius #267: PTG % in octoprint different from % in API

s31teg, 2021-09-29:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

**Describe the bug**
The % shown In the ocotprint dash is correct but the % fed back via the API progress is incorrect.

**To Reproduce**
Look at progress result via the API and compare to % in dash

**Expected behavior**
Both % to be correct

**OctoPrint Version:**
1.6.1

**PrintTimeGenius Version (if you know):**
2.2.8

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename the files).  **This is really important.  I probably can't help you without this.**

AlbertMN, 2021-10-18:
@eyal0 Hey Eyal :) Think the plugin in question is ours, "SimplyPrint-Cloud" (for SimplyPrint.io). We briefly emailed a while back.

We simply get the "time left" via. the OctoPrint plugin API, but if we missed a way to get the PTG progress instead, please do let us know how we can get the right one.

AlbertMN, 2021-10-19:
@eyal0 that helps a lot, thanks!

Just to be sure, when you're saying
> _"OctoPrint is replacing the web API's original function with this new one, only when the option is selected."_

Which option are we talking, and is OctoPrint _going_ to replace it? Or am I just missing something here?