eyal0/OctoPrint-PrintTimeGenius #269: Python 3.10 compatibility

cp2004, 2021-11-02:
**Describe the bug**
```
2021-11-02 15:49:47,968 - py.warnings - WARNING - /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py:186: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated
```
In Python 3.10, this stopped working, which means that the line that calls it stops working as well.

To retain compatibility with Python 2.7, it needs to be wrapped up to cover both cases, example below:

https://github.com/OctoPrint/OctoPrint/blob/945825f03fbd2417a009dd33daf9f787aff60211/src/octoprint/util/__init__.py#L29-L33

**To Reproduce**
Steps to reproduce the behavior:
1. Read/run code, view deprecation warnings. Haven't tested what breaks, but I know in Python 3.10 it will break.

**Expected behavior**
No breaking 🙂 

**OctoPrint Version:** 1.7.1

**PrintTimeGenius Version (if you know):** Latest, checked in the default branch on GH.

This is probably low priority, but we have seen a few people installing Py3.10 (for example on Windows) and it breaks plugins. OctoPrint has full compatibility to Py 3.10 since 1.7.0 (in theory!) if you want to test against it.

eyal0, 2021-11-02:
Would you like to send a PR?

cp2004, 2021-11-02:
I can't test it, which is why I was hesitant to send a PR

eyal0, 2021-11-02:
Ah.  Well, when I get around to it, I can do it.  I'd use pyenv I think.

LMBernardo, 2022-04-13:
The changes in my fork as of commit ecb210a9fb58536aea70e24a20f408355ba70637 are functional for me with Python 3.10. If someone wants to test this on lower versions of Python I can do a PR.

Nathan22211, 2022-06-23:
cp2004, just as a side note. Octoprint dropped support for python 2.7 after 1.8 I'm pretty sure

maludwig, 2023-07-14:
Quick little PR to fix this, tested with pyenv, on python 3.2, 3.3, 3.9, and 3.10

https://github.com/eyal0/OctoPrint-PrintTimeGenius/pull/308

eyal0, 2023-08-02:
Fixed in https://github.com/eyal0/OctoPrint-PrintTimeGenius/pull/308

closed 2023-08-02 by eyal0