eyal0/OctoPrint-PrintTimeGenius #271: Print TimeGenius analyse no longer runs

FreunderNacht, 2021-11-30:
**Describe the bug**

After uploading new files, the flashing golden star appears briefly.
However, this disappears again after some time.

The log says the following:

`2021-11-30 19:33:02,076 Not running built-in analysis.
2021-11-30 19:33:02,081 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode"
2021-11-30 19:33:02,082 Running: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode" ""
2021-11-30 19:33:05,681 Failed to run '"/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 337, in _do_analysis
    results_err = sarge_job.stderr.text
  File "/home/pi/oprint/lib/python3.7/site-packages/sarge/__init__.py", line 275, in text
    return self.bytes.decode(self.encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 273: invalid continuation byte
2021-11-30 19:33:05,683 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2021-11-30 19:33:07,039 Not running built-in analysis.
2021-11-30 19:33:07,057 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode"
2021-11-30 19:33:07,057 Running: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode" ""
2021-11-30 19:33:16,609 Failed to run '"/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 337, in _do_analysis
    results_err = sarge_job.stderr.text
  File "/home/pi/oprint/lib/python3.7/site-packages/sarge/__init__.py", line 275, in text
    return self.bytes.decode(self.encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 278: invalid continuation byte
2021-11-30 19:33:16,611 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2021-11-30 19:38:23,888 Not running built-in analysis.
2021-11-30 19:38:23,892 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode"
2021-11-30 19:38:23,893 Running: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode" ""
2021-11-30 19:38:27,522 Failed to run '"/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 337, in _do_analysis
    results_err = sarge_job.stderr.text
  File "/home/pi/oprint/lib/python3.7/site-packages/sarge/__init__.py", line 275, in text
    return self.bytes.decode(self.encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 273: invalid continuation byte
2021-11-30 19:38:27,523 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Boden_PETG.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2021-11-30 19:38:28,956 Not running built-in analysis.
2021-11-30 19:38:28,957 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode"
2021-11-30 19:38:28,958 Running: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode" ""
2021-11-30 19:38:38,503 Failed to run '"/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode" ""'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 337, in _do_analysis
    results_err = sarge_job.stderr.text
  File "/home/pi/oprint/lib/python3.7/site-packages/sarge/__init__.py", line 275, in text
    return self.bytes.decode(self.encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 278: invalid continuation byte
2021-11-30 19:38:38,505 Disabled: "/home/pi/oprint/bin/python3" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Gehause/LCD Display PCB/Gehäuse LCD Display PCB - Deckel_PETG.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
`
OctoPi and OctoPrint have been restarted several times without success.

**To Reproduce**

I think the problem occurred after the last update of the OctoPi.

**Expected behavior**
A clean analysis of the uploaded components.

**OctoPrint Version:**
Version 1.7.2

**OctoPi Version:**
Version 0.18.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

**PrintTimeGenius Version (if you know):**
The latest version.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename the files).  **This is really important.  I probably can't help you without this.**
[systeminfo.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/7628062/systeminfo.txt)
[plugin_PrintTimeGenius_engine (2).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/7628073/plugin_PrintTimeGenius_engine.2.log)

eyal0, 2021-11-30:
Looks like some sort of unicode issue?  I see mention of UTF in there.  Could be a problem with the version of python (python3 introduced some complications) or it could be due to non-ascii characters in the file/filename?  I see that you have umlaut?

Did it used to work okay for you?

FreunderNacht, 2021-11-30:
@eyal0 
Hi,
I haven't noticed any problems with the umlauts so far. I have already tried the whole thing with different files. There were also some that did not have umlauts in the file name.
However, I also had the error here.
So I assume that it is not due to the umlauts.
But I will still avoid the umlauts in the future.

FreunderNacht, 2021-11-30:
I tried to analyse an older file again.
This worked, whereas newer files do not:

[plugin_PrintTimeGenius_engine (5).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/7628855/plugin_PrintTimeGenius_engine.5.log)

I also noticed that at some point I had uploaded files with a diameter sign:

[image]

However, these seem to be still stuck in analysis.
Or what does the hourglass at the end mean?

eyal0, 2021-12-01:
The hourglass means that processing is still in progress.  It should become a star when it is done.  If it takes very long, either the file is big or there has been a crash.  Maybe the diameter sign is the problem?  Can you try to rename a file and see if it makes a difference?

FreunderNacht, 2021-12-08:
I have completely deleted the files with the special character (Ø) from Octoprint.
Now it seems to work again.
I also noticed that I had uploaded a file with an umlaut (ü), which was not analysed.
I then deleted it from Octoprint and converted the ü to a ue on my PC.
When I loaded the file back into Octoprint, it was updated without any problems.