eyal0/OctoPrint-PrintTimeGenius #273: wrong layer hight when PTG aktivated

Sekisback, 2022-01-08:
Hey eyal,

due to broken Octoprint I had to install everything new. Now I have a strange behavior with PTG.
I installed:

- DisplayLayerProgress Plugin: 1.27.2
- Dashboard Plugin: 1.19.5

till here i got the right maxZ in .metadata and in the state.

```
{
  "Unbenannt.gcode": {
    "hash": "ac9c636ea5fb52935674f11cae8b25d651de6459",
    "analysis": {
      "printingArea": {
        "maxX": 130.5,
        "maxY": 200.0,
        "maxZ": 20.0,
        "minX": 2.1,
        "minY": 20.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 180.0,
        "height": 20.0,
        "width": 128.4
      },
      "estimatedPrintTime": 1340.506435934754,
      "filament": {
        "tool0": {
          "length": 1329.4006599999987,
          "volume": 0.0
        }
      }
    },

```
If I install Print Time Genius Plugin: 2.2.8 and upload the same test gcode again, the .metadata are as follow,
 maxZ,  and also the height increase +1 LayerHeight 

```
{
  "Unbenannt.gcode": {
    "hash": "ac9c636ea5fb52935674f11cae8b25d651de6459",
    "analysis": {
      "analysisPending": false,
      "estimatedPrintTime": 1677.8810995454899,
      "printingArea": {
        "maxX": 130.5,
        "maxY": 200.0,
        "maxZ": 20.200000762939453,
        "minX": 2.0999999046325684,
        "minY": 20.0,
        "minZ": 0
      },
      "dimensions": {
        "width": 128.40000009536743,
        "depth": 180.0,
        "height": 20.200000762939453
      },
      "filament": {
        "tool0": {
          "length": 1322.4006347805262,
          "volume": 3.18074627886125
        }
      },

```
if I switch back to Print Time Genius Plugin: 2.2.7 the height in the state is shown right, but in the Dashboard the Layerheight will be +1
I dont know exactly where it comes from

env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.version: 1.7.2

eyal0, 2022-01-11:
Maybe Octoprint has changed the calculation of those values?

When PTG is installed, it replaces the calculation that OctoPrint is doing.  That is because the PTG calculation is faster and more accurate.

If you send me the file, maybe we can look inside the file to see what the correct answer must be?

Sekisback, 2022-01-11:
hey eyal, thank you that you will take time. uploaded a fresh testcube and the metadata, do you need something else
[metadata.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/7848153/metadata.zip)
[CalTube.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/7848155/CalTube.zip)
?

eyal0, 2022-01-11:
If you look at CalTube.gcode on line 8725, it is setting the Z to be 20.  Then on line 9013, relative positioning is turned on.  The on line 9015, 0.2 is added to the Z.  So the total Z is 20.2, which is the same as the max that PTG is reporting.

So I think that PTG is computing the max Z value correctly.

Can you explain to me the other problem that you are seeing?

Sekisback, 2022-01-11:
damn eyal you found it, thanks a lot you saved my day. Funny, I used the original "end gcode" from cura. I removed the z0.2 and now it looks fine. The other thing will be changed by Jefferey in the dashboard plugin. At the first look it seams that the height from the state and the dashboard were different, but this was a rounding error in the dashboard. The height was rounded after the first digit, if you use layer height 0.16 the fault height occurs after the 3rd layer

closed 2022-01-11 by Sekisback