eyal0/OctoPrint-PrintTimeGenius #275: FreeBSD Marlin-calc.amd64 not found

victor7376, 2022-01-28:
**Describe the bug**
Trying to print from Cura with PTG installed on OctoPrint causes Cura to wait for analysis to complete by PTG.  OctoPrint is installed in a jail on FreeNas / TrueNas which runs Freebsd.

**To Reproduce**
Steps to reproduce the behavior:
Install OctoPrint within a Freenas / Truenas Jail which runs off FreeBSD OS

**Expected behavior**
Cura to expect OctoPrint to complete analysis and start to print.

**OctoPrint Version: 1.7.3**

**PrintTimeGenius Version (if you know): 2.2.8**

As the calculator is not available for FreeBSD as it looks for marlin-calc.amd64, it makes Cura hang on waitingg for analysis to complete & you have to click on Start Print.

Log shows:

2022-01-27 19:12:03,548 Disabled: "/root/OctoPrint/venv/bin/python3.8" "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/.octoprint/uploads/CE3_Extra_Printable_Companion_Cube.gcode"
2022-01-27 19:12:03,548 Running: "/root/OctoPrint/venv/bin/python3.8" "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/.octoprint/uploads/CE3_Extra_Printable_Companion_Cube.gcode" ""
2022-01-27 19:12:04,052 Failed to run '"/root/OctoPrint/venv/bin/python3.8" "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/.octoprint/uploads/CE3_Extra_Printable_Companion_Cube.gcode" ""'
Traceback (most recent call last):
  File "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/__init__.py", line 339, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.amd64" "/.octoprint/uploads/CE3_Extra_Printable_Companion_Cube.gcode"
Can't find: /root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.amd64

eyal0, 2022-01-28:
Can you check if that file exists somewhere on your computer?  And if so, adjust the path in the PTG config?

victor7376, 2022-01-28:
I have just checked and there's no marlin-calc.amd64 anywhere in the jail, there's only those that came with PTG.

eyal0, 2022-01-28:
When PTG installs itself, it also should install marlin-calc....

Ah, I see, amd64 is not a supported platform because I didn't know about it!  This should be easy to solve.

can you check if you have any files at all that start with the name "marlin-calc"?  Maybe `/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.i686` for example?  In that directory, you will find many different marlin-calc executables.  Please test them until you find one that works when you run this:

    "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.amd64" "/.octoprint/uploads/CE3_Extra_Printable_Companion_Cube.gcode"

Let me know how that goes.

victor7376, 2022-01-28:
I have found the directory where the calc are:

/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/

Can you give any advice how to do the tests?  Is it via ssh or via the plugin itself?

Cheers

eyal0, 2022-01-28:
There are a few ways that you can do it depending on how comfortable you are with unix.

You could type this in:

    "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.x86_64" "/.octoprint/uploads/CE3_Extra_Printable_Companion_Cube.gcode"

And see what happens.

Or, you could go into PTG config and change the configuration in there, swapping `amd64` for `x86_64`.  I think that it's in the config there, right?

victor7376, 2022-01-28:
The config on octoprint just says

`"/root/OctoPrint/venv/bin/python3.8" "/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "{gcode}" "{mcodes}"`

I did try via ssh on all of them and only the windows exe was different.

The first few came up with errors: 

Exec format error. Binary file not executable.

Windows exe came up with:

```
/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.windows-x86.exe: MZ▒▒▒▒@▒▒▒: not found
/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.windows-x86.exe: 1: Syntax error: Unterminated quoted string
/root/OctoPrint/venv/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.windows-x86.exe: 2: Syntax error: Error in command substitution
```

The only one that is working at the moment that is NOT marlin is the 'All gcode analyzers (usually not as good as marlin-calc)' which uses a python script from what I see.

I did look through here to see if I could build a calc for amd64 but not sure how to go about it.

eyal0, 2022-01-28:
Aha, that's interesting!  I see!

You tried the x84_64 one and the i686 one, too?  And those didn't work?  Very interesting.

I guess that I'd need to cross-compile for your platform.  But I don't know what your platform is!

Can you tell me more about your computer?  It's unix?  Is it an Intel CPU?

Are you familiar with compiling code?  Would you like to try that?  If you have compiled code before then it's fast to do.

victor7376, 2022-01-28:
If it involves scon I have tried it but it throughs up errors too saying certain things cant be found.

I'm running OctoPrint in a jail which runs freebsd all running on True Nas Core:

This is the guide I used to install it:

https://community.octoprint.org/t/setting-up-octoprint-on-freenas-11-3-server-inside-iocage-jails/7440

I just changed the python version to 3.8.

root@Octo:/ # python3.8 -c "import platform; print(platform.system()); print(platform.machine())"
FreeBSD
amd64
root@Octo:/ #

The cpu: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz

eyal0, 2022-01-28:
The scons is trying to compile every platform but you only need one.  You can try to just compile only the one that you need.

```
git clone --branch estimate-bugfix-2.0.x https://github.com/eyal0/Marlin.git
cd Marlin/Marlin
scons ./out/x86_64/marlin-calc.x86_64
```

Let me know if that succeeds.

victor7376, 2022-01-28:
Error on running the command:

root@Octo:/Marlin/Marlin # scons ./out/x86_64/marlin-calc.x86_64
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
g++ -o out/x86_64/gcode/config/M200-M205.o -c -Wall -static -std=gnu++11 -O2 src/gcode/config/M200-M205.cpp
sh: g++: not found
scons: *** [out/x86_64/gcode/config/M200-M205.o] Error 127
scons: building terminated because of errors.
root@Octo:/Marlin/Marlin #

eyal0, 2022-01-28:
Ah, no compiler!  Well, let's install a compiler!

```
sudo apt-get install g++
```

victor7376, 2022-01-28:
Got it done, I think I've got it sorted out.

I had to install gcc with the command:

pkg install gcc

Then I ran the command you gave and it compiled it, I then copied it to the analysis folder and renamed it to marlin-calc.amd64  then ran the analysis from within the plugin and its now showing a star when it used to just show a egg timer.

victor7376, 2022-01-28:
Confirmed its now fully working.  Thank you

eyal0, 2022-01-28:
Just go here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/tree/master/octoprint_PrintTimeGenius/analyzers

Then click to upload a file:

[image]

Easy.

victor7376, 2022-01-28:
lmao I was just typing out how do you do that

victor7376, 2022-01-28:
err... Uploads are disabled.
File uploads require push access to this repository.

eyal0, 2022-01-28:
Ah.  Okay, you'll need to fork.

[image]

Click fork, make your own fork, add your file there, then make a pull request.

eyal0, 2022-01-28:
You're not going to break anything or screw anything up by doing this.  Don't be worried.  I will confirm that the PR looks good before I merge it so there's no danger that PTG gets screwed up for anyone.

victor7376, 2022-01-28:
added and sent ;)

PR #276 (2022-01-28): Adding file for FreeBSD
Adding file for freebsd installs of OctoPrint via Jails on FreeNas & TrueNas

This fixes #275
closed 2022-01-28 by eyal0