eyal0/OctoPrint-PrintTimeGenius #280: Possible to Create Your Own printer.cfg?

Torrax, 2022-03-07:
I have a FlashForge Creator Pro and was looking to use this plugin for better-estimated print times however the M503 code does not work on my printer so I was hoping that I could possibly set up the printer configuration manually. 

When I just run this with Marlin Calc it actually does a pretty good job of giving a very accurate estimation on print times already however it seems to mess with the filament usage which is used by other plugins such as CostEstimator.

I have attached a copy of the normal meta.json and the marlin meta.json and you can see some strange discrepancies between the two. I'm assuming this is due to the printer configuration not properly being set up. Is there some way I can do this manually?

[marlin.json.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8194439/marlin.json.txt)
[normal.json.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8194440/normal.json.txt)

eyal0, 2022-03-07:
Wow, it seems really off!  That maxX is wildly different!  I don't know why.  Which one is right?

The filament usage seems way off, too.  This might be due to PTG's marlin estimator (`marlin-calc`) thinking that the printer is in absolute extruding mode instead of relative extruding mode.  These commands are `M82` and `M83`.  I think that the default is absolute and in gcode files you can see that the E coordinate is always increasing.  Is your file like that?

You might also send a gcode file and the plugin log.  Those will help me debug this.

eyal0, 2022-03-07:
And to answer your question: Yes, you can put the config into the octoprint manually.  Look for this file:

`~/.octoprint/data/PrintTimeGenius/printer_config.yaml`

and modify the contents.  Here is what mine looks like:

```
printer_config:
- M92 X80.0 Y80.0 Z800.0 E90.0
- M203 X500.0 Y500.0 Z5.0 E25.0
- M201 X2000.0 Y2000.0 Z10.0 E74.0
- M204 S300.0 P750.0 R1000.0 T300.0
- M200 S0 D1.75
- M900 K0.01
- M205 B20000.0 E5.0 S0.0 T0.0 X10.0 Y10.0 Z0.3 J0.0
version: 2.2.7
```

You should do this while OctoPrint is not running so that PTG and your edits will not collide.

Each line in that list is processed by `marlin-calc` as if they were the first lines of every gcode file that will be processed.

The lines come from PTG "listening" to and intercepting:
* the output of `M503` and
* the typed input of the user.

I hope that this helps you.  Reach out if you need more help.

Torrax, 2022-03-08:
Thanks for getting back so quick! So I took a look into this and temporarily put your printer config in there for now to play around with. It looks like Flash Forge does not recognize a lot of the printer information commands.  Just to keep it consistant I removed it for the log files.

Here is a copy of PTG logs and Octoprint Logs as well as updated metadata files and the stl being tested gcode. The first entry in PTG logs is running marlin-calc the second is running the Octoprint built-in Analyser which shows a correct filament length and build area.

[four_square_cons.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8201661/four_square_cons.gcode.txt)
[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8201665/octoprint.log)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8201666/plugin_PrintTimeGenius_engine.log)
[PTG-Marlin.json.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8201688/PTG-Marlin.json.txt)
[OctoAnalysis.json.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8201690/OctoAnalysis.json.txt)

I was also able to play around with the .metadata file and correct the output for filament length and print size and it looks like that fixes the issue breaking the other plugins. Checking the G code it does look like it does use relative coordinates for positioning (M83). If PTG is thinking its absolute maybe that is the issue? Is there a way I can check/change what TPG uses?.

[FixedMarlin.json.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8201769/FixedMarlin.json.txt)

eyal0, 2022-03-08:
The `.metadata` file is regenerated everytime that you analyze a file.  So your modification will be temporary only.

marlin-calc doesn't support M83 and I never added support because almost no one uses it.  You could turn it off in prusaslicer?  I guess that I could also add support for it but that will take longer.

There are like 10 files attached to this ticket now so it has become a real pain to figure out what PTG is doing, what OctoPrint is doing, and what it ought to be doing.  So let's try to sort this all out!

Here is what OctoPrint is reporting:

```
      "printingArea": {
        "maxX": 70.0,
        "maxY": 16.609,
        "maxZ": 18.0,
        "minX": -121.0,
        "minY": -74.0,
        "minZ": 0.0
      },
      "dimensions": {
        "depth": 90.60900000000001,
        "height": 18.0,
        "width": 191.0
      },
      "estimatedPrintTime": 1053.0,
      "filament": {
        "tool0": {
          "length": 0.0,
          "volume": 0.0
        },
        "tool1": {
          "length": 275.1474999999987,
          "volume": 0.6393112492977976
        }
      }
```

Do those numbers seem correct?

Here's what PTG is reporting:

```
Analysis: {"estimatedPrintTime": 1054.90429939646105595, "printingArea": {"maxX": 16.11400032043457031, "maxY": 16.60600090026855469, "maxZ": 18.00000000000000000, "minX": -121.00000000000000000, "minY": -74.00000000000000000, "minZ": 0.20000000298023224}, "dimensions": {"width": 137.11400032043457031, "depth": 90.60600090026855469, "height": 17.79999999701976776}, "filament": {"tool1": {"length": -1.09999972244258970, "volume": -0.00264580939534135}}}
```

Right?

eyal0, 2022-03-08:
The `maxX` is different!  PTG is saying 16.11, OctoPrint is saying 70.0.  Which is correct?

eyal0, 2022-03-08:
On further inspection of your gcode, it's not clear to me how it works at all!  There's some weird stuff going on in there.

Are you able to print this gcode successfully?

I'll explain.  If you look at the docs for [G90](https://marlinfw.org/docs/gcode/G090.html) and [G91](https://marlinfw.org/docs/gcode/G091.html), you'll see that they set the relative/absolute positioning mode for *all* the axes, that is, X, Y, Z, and E.

Now if you look at the docs for [M82](https://marlinfw.org/docs/gcode/M082.html) and [M83](https://marlinfw.org/docs/gcode/M083.html), you'll see that they set the relative/absolute mode for just the `E` axis.

So what happens when M82/M83 appear before G90/G91?  It seems that the latter overrides the former according to the docs. 
 But it seems like you are getting a different behavior!

In the code for Marlin, I see that the code which sets E to be relative with M83 is not doing the same thing as how E is set to relative with G91!  This is completely crazy.  I do not understand it.

As far as I can tell, your gcode that you sent should not print properly.

eyal0, 2022-03-08:
Ah wait, however, the way that the code works is that there is a function `axis_is_relative` that does *not* examine the bit set by G90, no!  It looks at the bits set by M92/M93.  Oy.  This is going to be difficult.  I will raise an issue with the Marlin people.

eyal0, 2022-03-08:
Ah wait, do you have more than one extruder?

NeverDarktan, 2022-03-10:
The Flash Forge Creator is a dual extrusion printer.

eyal0, 2022-03-10:
I may have properly supported dual extruders which is why the volume is wrong. This needs to be fixed in marlin-calc.

As a work around for now, use absolute extruder position mode. This may be an option within PrusaSlicer.

Torrax, 2022-03-11:
Sorry it took so long to reply. You had some interesting input and got me troubleshooting a bunch of things and made some good progress. 

So the Flash Forge Creator Pro uses Sailfish, a very weird kind of gcode so you don't see many people use and I didn't know how the marlin-calc would work on but it seemed to give super accurate results for time but was messing up other parts of the metadata. 

I tried switching to absolute extrusion mode my prints began failing and it seems the profiles I have found online for this printer seem to force relative extruding in the code so I'm not sure if that was supported by my printer.

On Prusa I was able to find an alternative G-Code Flavor by Makerbot that works with my printer and the Filament Used started working again. The weirdest part about this however is the g-codes are identical.
I have attached copies below: (MakerWare - Working | Sailfish - Not Working)

[four_square_(MakerWare).gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8228373/four_square_.MakerWare.gcode.txt)
[four_square_(Sailfish).gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8228374/four_square_.Sailfish.gcode.txt)

The only issue is estimated time all of a sudden was extremely off.  Fortunately, you did an extremely good job programming this and PTG will actually adjust its time based on how close it was to actual print time. After running it for a bunch of test prints  I was able to get it accurate within 3% of the accurate time which is perfect for me!

As for the print size, it turns out Octoprint and PTG were both getting it wrong, the print is only a 15x15mm print. I never noticed the print size was so off as well but it never affected much.

My printer is a more specific case scenario and I don't want you to make any changes for that. The majority of the issues seem to have been fixed somehow.

For the others having issues with CostEstimation and TPG try playing with different settings in your gcode.

closed 2022-03-11 by Torrax

eyal0, 2022-03-11:
Glad that it started working!

If you want to tune the compensation even more, you can try this proposed change:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/214#issuecomment-1051398573