eyal0/OctoPrint-PrintTimeGenius #283: automatic analysis not working

derekakelly, 2022-05-03:
Hello, 
I setup my server to have an shared network folder for octoprint. I save directly from my slicer to this folder. After hitting refresh in the octoprint files list, it comes up as expected. But even after loading the file, and/or preheating, the automatic analysis doesn't work. In order to get my gold star, I'm forced to go into the settings and manually run it.

I expected the analysis to happen either after it was loaded, or as it was preheating.

Ubuntu Server 21.10
OctoPrint Version 1.7.3
PrintTimeGenius Version 2.28
[plugin_PrintTimeGenius_engine (1).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/8614209/plugin_PrintTimeGenius_engine.1.log)

Thanks for your time!

-Derek

derekakelly, 2022-05-03:
I just ran a short print as a test. It's also not performing the analysis while it's printing. So I printed it a second time, and it still hasn't analyzed it yet.

eyal0, 2022-05-04:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/112#issuecomment-451156530

Please read that and try again. If it doesn't solve your problem, reopen this issue.

closed 2022-05-04 by eyal0

derekakelly, 2022-05-07:
Although you were right, I didn't have the watch folder properly setup and I was uploading to the upload folder rather than the watched folder. Since changing those settings and letting it run for a few days, I unfortunately haven't noticed a difference.

So I adjusted my logging settings and I'll let it run for a few more days and see what kind of information I can gather.
[image]
Should I add anything else?

derekakelly, 2022-05-08:
After further testing, I restarted octoprint and that was the silver bullet. The folder changes began to work as expected and files are now automatically being processed. Thanks for your help!

eyal0, 2022-05-09:
Glad that it got sorted out!  Sorry for the trouble.  The Watched folder, I feel, is one of major feature of OctoPrint that is not well described.  Enjoy!