eyal0/OctoPrint-PrintTimeGenius #288: Genius is not running

DaBigPenguin, 2022-09-18:
Print time genius has stopped analyzing files. Setting to debug level and the log contains nothing.  The uninstall does not function and "./oprint/bin/pip install install --user --force-reinstall "https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip"" does not force a reinstall.

 

**OctoPrint Version:**
OctoPrint 1.8.2
    Python 3.7.3
    OctoPi 0.18.0

**PrintTimeGenius Version (if you know):**
2.28

eyal0, 2022-09-19:
Are there no error messages anywhere?  Perhaps read the logs of Octoprint?

eyal0, 2022-09-19:
I'll close this issue because the bug is in a different program and not this one.

closed 2022-09-19 by eyal0

DaBigPenguin, 2022-09-19:
Removing slicer estimator fixes Print time genius.