eyal0/OctoPrint-PrintTimeGenius #289: Restoring Python 3.7 based backup to Python 3.9 based image causes PrintTimeGenius to fail

n8many, 2022-09-19:
**Describe the bug**
After migrating my system to a new Pi and updating to the newest version of Octopi with it, I could not get accurate time estimates or filament usage in Octoprint. I later tracked down the issue to PrintTimeGenius failing to run, with the issue being it was pointing to the old version of the python venv (python3.7) rather than the new one (python3.9).

I did have to manually call the restore function via command line, but I am not sure if that would have changed the result since the issue seems to be how the settings are saved.

**To Reproduce**
Steps to reproduce the behavior:
1. Backup Octoprint configuration from Octopi image <= 0.18.0 with PTG installed.
2. Reflash Pi to OctoPi v1.0.0 (RC2 is current release).
3. Restore Octoprint configuration.
4. Upload new file and PTG will fail to run with the following message:
```
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_PrintTimeGenius/__init__.py", line 339, in _do_analysis
    raise Exception(results_err)
Exception: /home/pi/oprint/bin/python3: can't open file '/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py': [Errno 2] No such file or directory
```

**Expected behavior**
I would expect the paths that PTG saves would be agnostic to where the "oprint" python venv is located. Perhaps there is a way to save the command with some sort of variable that changes based on the location of the venv or simply uses relative paths.

Manually updating the command string to read python3.9 fixes the issue and allows the plugin to run.

**OctoPrint Version:**  1.8.2  (OctoPi 1.0.0 RC2, unsure if this is only an octopi thing)

**PrintTimeGenius Version:** 1.28.0

eyal0, 2022-10-03:
The config stores the location into it.  You can see it in the configuration for PTG.  If the configuration for PTG is edited, then the edit will override the defaults and the new defaults won't take affect when you update.

I don't know if there's a way for me to fix this.  If you have an idea, re-open this issue.  If this happens a lot, we can try to think of something smarter like maybe a script that is in a static location but knows how to search for the correct version of python.

mfraser, 2023-05-20:
Came across this issue when upgrading to 1.9.0, wondered if you could use https://note.nkmk.me/en/python-sys-platform-version-info/ to set that part of the location?