eyal0/OctoPrint-PrintTimeGenius #294: Meat Pack and Arc Welder

stonemandave, 2022-10-28:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

**Describe the bug**
Estimations are way off when using meatpack and arc welder.

**To Reproduce**
Steps to reproduce the behavior:
1. Activate Meatpack in marlin.
2. Activate Meatpack and Arc Welder plugin
3. Start Print
4. See error

**Expected behavior**
I won't know why it can't estimate geode but I didn't expect it to be off by over half the actual duration.

**OctoPrint Version:**
Newest
**PrintTimeGenius Version (if you know):**

Any print will replicate the error no logs nor geode needed. I did a basic cylinder calibration from cura.

eyal0, 2022-10-28:
Yeah, I think that there might be some issues with plugins that modify the gcode.  So here's the thing, you say that you did both MeatPack and ArcWelder and now PrintTimeGenius doesn't work.  But can't tell if the problem is ArcWelder or MeatPack because you did both at the same time.

Is it possible for you to do just one or just the other and then check?

My guess is that the ArcWelder is the issue.  I think that I've seen problems around that before.  It might be that I'm not correctly calculating for arcs.  I can fix it but I'll need a test file to look at.

If you could, please send the file after it is modified by ArcWelder.  Also, send the time that is actually took to print the file along with the time that you expected.  Also please send the PrintTimeGenius log that includes the analysis of the gcode file so that I can see what it was doing.  Thanks.

stonemandave, 2022-10-28:
Thank you I will get all that soon

stonemandave, 2022-10-29:
Couldn't replicate i guess it compensated for it in the end.

closed 2022-10-29 by stonemandave

eyal0, 2022-10-29:
Good to hear! Enjoy!