eyal0/OctoPrint-PrintTimeGenius #301: Auto analysis doesn't run if files are manualy added to uploads folder

mlewis-everley, 2023-03-30:
**Describe the bug**
I am not 100% sure this is a "bug", but if you add a gcode file to the octoprint uploads folder outside of octoprint, PrintTimeGenius sees the file, but you have to manually analyse the file (via the plugins menu).

I am running several clones of the same printer, so it is easier to export the gcode to one location and then rsync it to each printer's octoprint install. But doing this means that you need to manually analyse the gcode.    

**To Reproduce**
1. Generate a gcode file
2. Add it to octoprint uploads folder manually
3. Octoprint sees the file, but print time genius doesn't auto analyse

**Expected behavior**
I guess it would be nice if the plugin was able to be configured to analyse any unchecked files automatically? Would this have to be some form of cron job though?

eyal0, 2023-03-30:
This is #283 . Try the watched folder. Reopen this issue if that doesn't solve your problem.

closed 2023-03-30 by eyal0

mlewis-everley, 2023-03-30:
Thanks for the link, I did look for existing issues but somehow managed to miss this one :-/

I will give it a go, hopefully #283 will be the answer I need :-)