eyal0/OctoPrint-PrintTimeGenius #302: Percentage shown "12.96omplete" not "12% complete" or "12.69% complete"

tehniq3, 2023-03-31:
On ANET A8 LCD 2004 display:
[image]

I think a INT variable is enough, but is just my oppinion !

eyal0, 2023-03-31:
Is this a bug in PTG?  When you disable PTG, does this bug go away?

tehniq3, 2023-03-31:
if not use your plugin, on last row of display printer not show any info, but if printer is not connected to Octoprint show "Printing  12%"

tehniq3, 2023-04-01:
I think that is an issue from other plugin ... sorry for my mistake.. you can close the post...

tehniq3, 2023-04-01:
sorry for my mistake, other plugin is responsable for issue !

closed 2023-04-01 by tehniq3

eyal0, 2023-04-01:
I can probably guess the reason that you are having a problem.

In the other plugin where it says "%" in the code or in the configuration, it should say "%%". The "%" is special and it must appear twice.

tehniq3, 2023-04-02:
you are right ! instead one % I put %%% 😎 I need to change in {completion:.2f}%%% done to see "16.4% done"
[image]
PS: many thank to you... nota some beer from my side ! 🤘🍺🍺🍺