eyal0/OctoPrint-PrintTimeGenius #309: Migration to Mainsail / Klipper

PatrickGlatz, 2023-08-12:
Hi @eyal0 

i loved your Plugin while using OctoPrint. In the meantime, i have switched to Mainsail for several reasons and one of the two things i am only missing is your PrintTimeGenius Plugin. With my configuration of Cura and/or Klipper, i do have errors up to Factor 2 which was reduced to few minutes only when using your plugin.

Would you think about a migration to Mainsail? I really would love to see that and i swear, there are thousands of other users, which would also do. I would also be a beta tester in case you need some.

I am looking forward for a ping of you.

Thanks in advance
Patrick

eyal0, 2023-08-21:
I haven't used a 3D printer in a decade now.  Unfortunately, I don't even own one.  If someone wants to do this, I'm happy to assist.  I've never used Mainsail but I hear that it is gaining popularity.

I suggest that you find a Mainsail developer that would want to do this or knows someone that would want to do this.  And then put them in touch with me.

Peuqui, 2024-12-13:
I second that! 
PrintTimeGenius is the one and only, which is capable of computing the (nearly) correct time needed.
As I switched to Mainsail, too, I missed that plugin heavily. I am a hobbyist programmer, but unfortunately not in the world of Python, plugins, Raspi, etc. I would love to see your code either in Mainsail or OrcaSlicer!
Maybe, I start an issue thread there to push this forward... :-)

Thanks for your effort!
Markus