eyal0/OctoPrint-PrintTimeGenius #311: AttributeError: module 'collections' has no attribute 'Mapping'

flaviut, 2023-11-05:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

**Describe the bug**

My analysis failed due to the following error:

```
2023-11-05 11:33:21,133 - octoprint.printer.standard - ERROR - Error while pausing the analysis queue
Traceback (most recent call last):
  File "/opt/octoprint/lib/python3.11/site-packages/octoprint/printer/standard.py", line 1482, in on_comm_state_change
    self._analysisQueue.pause()  # only analyse files while idle
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octoprint/lib/python3.11/site-packages/octoprint/filemanager/analysis.py", line 129, in pause
    q.pause()
  File "/opt/octoprint/lib/python3.11/site-packages/octoprint/filemanager/analysis.py", line 265, in pause
    self._do_abort()
  File "/opt/octoprint/lib/python3.11/site-packages/octoprint_PrintTimeGenius/__init__.py", line 207, in _do_abort
    if _allow_analysis(self._plugin._printer, self._plugin._settings):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/octoprint/lib/python3.11/site-packages/octoprint_PrintTimeGenius/__init__.py", line 186, in _allow_analysis
    if not isinstance(thermostat, collections.Mapping) or not 'actual' in thermostat or not 'target' in thermostat or thermostat['target'] is None:
                                  ^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'Mapping'
```

**To Reproduce**
Steps to reproduce the behavior:

Run this code on Python 3.10+, where the `collections.Mapping` field was moved to `collections.abc.Mapping`. `collections.abc.Mapping` was introduced in Python 3.3, released in 2012

**Expected behavior**

No error

**OctoPrint Version:**

Version 1.9.3

**PrintTimeGenius Version (if you know):**

2.3.1

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename the files).  **This is really important.  I probably can't help you without this.**

closed 2023-11-06 by eyal0

bereldhuin, 2024-03-08:
Hi,

Would it be possible to create a new tag with this bug solved ? 
I think most people installing it on newer Ubuntu version will have the problem with Python 3.10.

eyal0, 2024-03-09:
I don't understand, what am I supposed to do here?

flaviut, 2024-03-09:
@eyal0 https://stackoverflow.com/a/18223354/2299084, and create a new `2.3.2` version. It's been a while, so I don't remember, but maybe Octoprint's package manager requires a tagged version to update?

eyal0, 2024-03-10:
Ah, right, that kind of tag, yes!  Hah, I had forgotten.  I also need to update the version number to match in the code or else the updater will try to download the code over and over again!

eyal0, 2024-03-11:
e9489e63606f65daf6e15455c184a40158cb238e

eyal0, 2024-03-11:
Done, here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/releases/tag/2.3.2

Let me know if I need to do something else, thanks.

PR #312 (2023-11-06): Fix python 3.10 compatiblity
Fixes #311
commit 5c237c1c (2023-11-06): Fix python 3.10 compatiblity

Fixes #311