eyal0/OctoPrint-PrintTimeGenius #313: No analysis after first file jump

Felipe-NS, 2023-11-11:
**Describe the bug**
My workflow works like that: I send a file to the printer and directly print it(without whait for the PTG analysis. But..After that I usually send some other files just to make PTG analyse this file.
The problem when I try to do that the PTG plugin don't automaticaly read the file even with "Allow analysis while printing..." active

**To Reproduce**
Send file directly without analysis and then send other file.

**Expected behavior**
No analysis. Some times it work, not shure what happens in the background.

**OctoPrint Version:**
 1.9.3 

**PrintTimeGenius Version (if you know):**
 2.3.1 (but It always worked like that)

[plugin_PrintTimeGenius_engine (5).log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/13326758/plugin_PrintTimeGenius_engine.5.log)

eyal0, 2023-11-12:
Are you sending the file to the correct directory? Which one?

Felipe-NS, 2023-11-12:
I always use api/files/local

Felipe-NS, 2023-11-12:
It seems like the file that didn't completed analysis block other files in queue list.

eyal0, 2023-11-12:
You're supposed to put the file in the watched directory.

Any clues in the log file?

the-stag, 2023-12-16:
I'm having a similar problem, but I think it's related to the latest version of Prusa Slicer. Nothing I've sliced since the update to 2.7 seems to get analyzed. I'm not using binary gcode, but I know there are some other new features such as arc fitting with g2 and g3 moves. I haven't had a chance to test, but think it could be related.

eyal0, 2023-12-17:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/119#issuecomment-466131587 ?

eyal0, 2023-12-17:
I don't see any errors in the log file.

the-stag, 2023-12-18:
How can I manually upload the log?