eyal0/OctoPrint-PrintTimeGenius #318: Negative print time bug

AlbertMN, 2024-07-09:
**Describe the bug**
The "time left" reported by PrintTimeGenius shows as a negative number. I'm quite sure this is a PrintTimeGenius issue, as I found a very old bug report from OctoPrint, where Gina reports that it should no longer be possible to get negative print time in OctoPrint (https://github.com/OctoPrint/OctoPrint/issues/692).

**To Reproduce**
This is where it gets tricky... It was a customer of ours (I'm from SimplyPrint.io) that reported this, but for his setup, the steps to reproduce was quite simply starting the print.

I tried to do this on one of our Prusa MK3S printers, on OctoPrint 1.10.2, though, and didn't get this issue.

We've never had issues with this before, it's the first time I've seen this across thousands and instances, never been reported before, and I can find any mention of it in issues on this repository, or any mention of cases like this on the OctoPrint forum.

**Expected behavior**
Not getting negative print time

**OctoPrint Version:**
1.7.3

**PrintTimeGenius Version (if you know):**
2.2.8

Here you'll find a zip with the logs (OctoPrint main log, SimplyPrint (plugin) communication log, where you can see the negative print time, the PrintTimeGenius log) as well as the Gcode file _(wouldn't let me upload `.gcode` without it being in a .zip - sorry!)_;
[logs-and-file.zip](https://github.com/user-attachments/files/16149669/logs-and-file.zip)

eyal0, 2024-07-10:
On line 15 of the log file, we see reasonable values for the print time.  But on line 23, we see negative values.  These are caused by the compensation, which has the "average heat-up time" as 34658.462530549616 seconds.  This means that PTG has computed that it takes about 9.6 hours to heat up the print bed, which is clearly wrong.

The print time compensation values can be cleared so that they will be recalculated.  They can also be fixed to more reasonable values if the user wants to measure how long it takes to get from room temperature to printing temperature.  Clearing these compensation values should solve the problem.

More information here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems#printing-time-is-way-off

closed 2024-07-10 by eyal0

AlbertMN, 2024-07-10:
Hi @eyal0
Thanks for your response!

We did figure it may be the "is way off" issue, but since it went negative, we reported it just in case it was a new bug.

The user has been instructed to clear the compensation values. And now, if any user has a negative print "time left" estimate, they'll hopefully Google and find this issue! :)