eyal0/OctoPrint-PrintTimeGenius #319: No Apple Silicon support for Marlin simulation (darwin-arm64)

Hillshum, 2024-07-13:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

**Describe the bug**
The Marlin simulation is not available for Apple Silicon (darwin-arm64)

**To Reproduce**
Steps to reproduce the behavior:
When running on an Apple Silicon Mac, trigger a file analysis

**Expected behavior**
A clear and concise description of what you expected to happen.

**OctoPrint Version:**
2.0.0.dev2872+ga652a0ec2.dirty
**PrintTimeGenius Version (if you know):**
2.3.3

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename the files).  **This is really important.  I probably can't help you without this.**
Here's a snippet
```
Loads/CE3PRO_disney_frame-holder.gcode" ""'
Traceback (most recent call last):
  File "/Users/hiltonshumway/code/OctoPrint/venv/lib/python
3.9/site-packages/octoprint_PrintTimeGenius/__init__.py", l
ine 347, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/Users/hiltonshumway/code/OctoPrint/ve
nv/lib/python3.9/site-packages/octoprint_PrintTimeGenius/an
alyzers/marlin-calc.darwin-arm64" "/Users/hiltonshumway/Lib
rary/Application Support/OctoPrint/uploads/CE3PRO_disney_fr
ame-holder.gcode"
Can't find: /Users/hiltonshumway/code/OctoPrint/venv/lib/py
thon3.9/site-packages/octoprint_PrintTimeGenius/analyzers/m
arlin-calc.darwin-arm64

eyal0, 2024-07-16:
Okay!  This requires compiling for a different platform.  Usually, I cross-compile but I don't know if I will find the compiler that runs on Unix but generates code for darwin arm64.  I see a few that are `arm64` but I don't know if they will work for darwin.

Are you able to compile for darwin-arm64?  I can teach you how.  You would be compiling the `marlin-calc` source code on your computer.

Hillshum, 2024-07-16:
Yeah, I'm happy to compile (or do any necessary testing)

eyal0, 2024-07-18:
Okay, you'll need to clone the `marlin-calc` code to your darwin arm64 system and then compile there.  clone this repo: https://github.com/eyal0/Marlin

And then, in the Marlin directory, where there is already a file called sSConstructs, you need to run `scons`.  This should compile the code and create a folder called "out" with the result in there.  It will try to compile many versions and some of them might not work.  Hopefully one of them will!

After you're done, try running `marlin-calc` from in the `out` folder.  You can run it on a gcode file or just run it without a file.  If the compilation was successful, it won't print out any errors about architecture or linking.  When you get a successful compile, then we'll do more testing of it and then I'll put it into PrintTimeGenius.

That's the high-level overview.  Let me know if you need help with any of the steps.

Hillshum, 2024-07-20:
Okay, so I ran into the problem where I didn't have the crosscompilers installed myself. I commented out most of the configs in `SConstructs`, leaving in one for darwin. Running the analysis seems to work.

```
env = Environment(
    CXX='clang',
    CPPFLAGS= '-Wall -std=gnu++14 ' + debug_flags,
    LINKFLAGS='-Wall -std=gnu++14 -lstdc++ ' + debug_flags)
add_program(env, 'darwin-arm64')```

eyal0, 2024-07-22:
Cool!  If you feel that you have a version of `marlin-calc` that will work on darwin arm64, maybe just add it to the `PrintTimeGenius` repo in a new PR?  That should be all that is necessary to get PTG to work for you on your platform.

Send the PR and I'll take a look.

Hillshum, 2025-02-19:
Just added #324

eyal0, 2025-02-19:
I have no way to test this but I hope that you got it right!  Thanks.

closed 2025-02-19 by eyal0