eyal0/OctoPrint-PrintTimeGenius #320: PTG way off - huge compensation values

klillevo, 2024-07-25:
PTG used to be  very accurate but is now estimating more than double the actual print time. And the estimated time remaining fluctuates wildly while printing. Sometimes up, most of the time rapidly  decreasing.

I am not sure what triggered the issue, or exactly when it started behaving this way.

I notice rather large compensation values. 

MK3S+, PrusaSlicer 2.8.  MeatPack gcode compression

[image]

klillevo, 2024-07-26:
I deleted all the compensation values - and that seems to have fixed the issue for the next print, at least a very short one. That the compensation values had become so messed up is probably a bug. However, please feel free to close this issue if the behavior is kind of expected and the compensation values  need to be removed once in a while.

klillevo, 2024-07-27:
On the 2nd print after deleting compensation values, they had become very large, and now it seems PTG is unable to  estimate the print time. It just remains a dash and a red dot.  I am not sure what to try next. Remove and re-install probably.

eyal0, 2024-07-28:
Perhaps other plug ins are interfering with PTG?  Can you send a log file from PTG?

klillevo, 2024-07-28:
Okay, on another print I started now, it's back to being very accurate (after I enabled the log). I will upload the file next time the inaccuracy occurs. Thanks!

myrrien, 2024-08-25:
I have the same problem as OP

rtrout, 2024-11-17:
I've started getting this problem too. Only noticed it recently, not sure how long it's been happening and what might have triggered it :-(  I have been experimenting with plugins, so that may have been the trigger. As the OP described, PTG estimates almost twice Prusa slicer and actual.

Log attached. Ran M503 and I can see a gcode in the log. 
[plugin_PrintTimeGenius_engine (1).log](https://github.com/user-attachments/files/17788944/plugin_PrintTimeGenius_engine.1.log)

Let me know if there is more I can provide, I'm new to this.

eyal0, 2024-11-18:
From the file you sent, your first print of `PlayerMarkers_v4_0.2mm_PLA_MK3S_3h30m.gcode` shows that it was calculated to take 22678 seconds plus another 128 seconds to warm up the printer and another 12.5 seconds to cool down.  Are those accurate?

> Let me know if there is more I can provide, I'm new to this.

* What is the name of the file that you tried to print?
* How long did PTG say that it would take?
* How long did it actually take?

rtrout, 2024-11-19:
As indicated in the gcode filename, prusaslicer estimated at 3h30m, which is roughly correct. As you point out ptg calculated at 6hrs.

Not sure when this started happening, only that ptg is now off.

eyal0, 2024-11-19:
PTG ran assuming that your printer was configured like this:

```
M220 S50
M92 X100.00 Y100.00 Z400.00 E280.00
M203 X100.00 Y100.00 Z12.00 E120.00
M201 X960 Y960 Z200 E5000
M204 P1000.00 R1250.00 T1000.00
M205 B20000 E4.50 S0.00 T0.00 X8.00 Y8.00 Z0.40
M207 F2700.00 S3.00 Z0.00
M208 F480.00 S0.00
```

Is that how your printer is configured?  You can view your printer configuration by typing `M503` into the console.  You may have settings in addition to the ones above.  Only the ones above need to match.

rtrout, 2024-12-07:
This is what M503 reports for me. The feedrates seem to differ for stealth, which I don't use. Is there something I'm missing?
\

Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E280.00
Recv: echo:UStep resolution:
Recv: echo: M350 X16 Y16 Z16 E32
Recv: echo:Maximum feedrates - normal (mm/s):
Recv: echo:  M203 X200.00 Y200.00 Z12.00 E120.00
Recv: echo:Maximum feedrates - stealth (mm/s):
Recv: echo:  M203 X100.00 Y100.00 Z12.00 E120.00
Recv: echo:Maximum acceleration - normal (mm/s2):
Recv: echo:  M201 X1000 Y1000 Z200 E5000
Recv: echo:Maximum acceleration - stealth (mm/s2):
Recv: echo:  M201 X960 Y960 Z200 E5000
Recv: echo:Acceleration: P=print, R=retract, T=travel
Recv: echo:  M204 P1000.00 R1250.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (us), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E4.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P15.23 I1.27 D45.76
Recv: echo:PID heatbed settings:
Recv: echo:   M304 P126.13 I4.30 D924.76
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:   M207 S3.00 F2700.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:   M208 S0.00 F480.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:   M209 S0

rtrout, 2024-12-08:
This is what M503 reports. I don't understand to to tell PTG to 'reread' from M503 to make sure it is using current settings?

Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E280.00
Recv: echo:UStep resolution:
Recv: echo: M350 X16 Y16 Z16 E32
Recv: echo:Maximum feedrates - normal (mm/s):
Recv: echo:  M203 X200.00 Y200.00 Z12.00 E120.00
Recv: echo:Maximum feedrates - stealth (mm/s):
Recv: echo:  M203 X100.00 Y100.00 Z12.00 E120.00
Recv: echo:Maximum acceleration - normal (mm/s2):
Recv: echo:  M201 X1000 Y1000 Z200 E5000
Recv: echo:Maximum acceleration - stealth (mm/s2):
Recv: echo:  M201 X960 Y960 Z200 E5000
Recv: echo:Acceleration: P=print, R=retract, T=travel
Recv: echo:  M204 P1000.00 R1250.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (us), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E4.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P15.23 I1.27 D45.76
Recv: echo:PID heatbed settings:
Recv: echo:   M304 P126.13 I4.30 D924.76
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:   M207 S3.00 F2700.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:   M208 S0.00 F480.00

eyal0, 2024-12-09:
That's interesting, I didn't know about the stealth feature.

Every time your printer prints out that M503 info, PTG is listening and it will read all the info and store it internally.  That's how it is able to know how fast your printer can go.  You'll find that the commands printed there by the M503 match the commands in your log file.

PrintTimeGenius doesn't know about stealth mode and regular mode.  It just looks at the last one that printed out.  The stealth one is printing after the regular one so it is doing a stealth mode calculations.  Maybe if it did the regular calculations it would be correct?  Do you want to try to run marlin-calc with the regular mode parameters instead of the stealth mode and see if it actually computes 3h30m?

rtrout, 2024-12-15:
I'm guessing this may have been introduced by firmware 3.14.0 - the timing would be about right.

Dumb question, how do I run marlin-calc with the regular mode parms? Looking in PTG plugin settings didn't help me.

eyal0, 2024-12-15:
On the same computer that OctoPrint is running, which may be a Raspberry Pi, you log in to get a terminal window and that is where you run marlin calc. The exact parameters used for running marlin calc are listed in logfile for PTG. It'll be the marlin calc executable file, followed by optionally a list of M codes that configure marlin calc, and the file name to process.

rtrout, 2024-12-16:
Ok, thanks for the pointer, found the call to marlin-calc, and as you anticipated, it is using the stealth parms, presumably because M503 is report stealth mode parms AFTER normal mode parms. This would explain why PTG calcs are mis-matched.

So at this stage it seems my only option is to disable PTG, and rely on the PrusaSlicer settings? Or are there other options I'm blind to?

eyal0, 2024-12-16:
Do you often use stealth mode? You could set the stealth parameters to the same as the normal parameters and then PTG would work.

Alternatively, PTG could be altered to support stealth mode. But that's a lot of work for me and I'm probably not going to do it because I'm busy. I would be willing to support someone else that does it.

rtrout, 2024-12-17:
No problem, I can't expect you to make a change for me. I'm inquiring on Prusa forums as to how the multiple settings got there, will report here if I find anything in case it helps others.