eyal0/OctoPrint-PrintTimeGenius #322: PTG completely breaks uploads in OctoPrint 1.11.0rc1

foosel, 2025-01-29:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

**Describe the bug**
PTG monkey patches FileManager's `add_file` method. With 1.11.0rc1 this method has received a new keyword parameter `user` to track user uploads. The patched replacement method doesn't support this parameter and makes all uploads fail:

```
2025-01-29 08:10:00,811 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/server/util/flask.py", line 1643, in decorated_view
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/vendor/flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
         ^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/server/api/files.py", line 692, in uploadGcodeFile
    added_file = fileManager.add_file(
                 ^^^^^^^^^^^^^^^^^^^^^
TypeError: PrintTimeGeniusPlugin.on_startup.<locals>.new_add_file() got an unexpected keyword argument 'user'
```

Monkey patching is something I don't consider that big of a problem, I do it myself when I don't have another choice, but you have to be *very* careful when you do it with internals - especially undocumented ones - and ideally work heavily with `*args, **kwargs` in your overwritten method signatures, to protect against issues like this one by always staying compatible to added parameters.

I fear I will have to blacklist PTG for OctoPrint 1.11.x until this is fixed as it is effectively rendering OctoPrint unusable for its purpose 😕

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade OctoPrint to 1.11.0rc1
2. Try to upload a file
3. See error

**Expected behavior**
I'd expect uploads to work when PTG is installed.

**OctoPrint Version:** 1.11.0rc1

**PrintTimeGenius Version (if you know):** 2.3.3

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename the files).  **This is really important.  I probably can't help you without this.**

I don't think the log will be of any help here as the error happens within OctoPrint, but here you go:

[plugin_PrintTimeGenius_engine.log](https://github.com/user-attachments/files/18584132/plugin_PrintTimeGenius_engine.log)

Also here's a very basic file with which to reproduce after renaming back to `.gcode` (any gcode works though): 

[upload.gcode.txt](https://github.com/user-attachments/files/18584139/upload.gcode.txt)

foosel, 2025-01-29:
I've provided a fixing PR: #323

closed 2025-01-30 by eyal0

foosel, 2025-01-30:
Can you tell me in which version this will be fixed? I can then modify the blacklist to allow loading of PTG again from that version onwards.

eyal0, 2025-01-30:
I'll just do it right now!  Version 2.3.4 has your fix.

PR #323 (2025-01-30): fix: fix uploading on OctoPrint 1.11+
Extends `new_add_file`'s signature by `*args, **kwargs` to no longer cause errors on added parameters for this function.

Fixes #322 

Note: I found the comment above that monkey patched function quite amusing given that it said to remove the monkey patch after a specific PR was merged - said PR has been merged now for over 7 years. I thus changed the TODO to check if that patch can now be removed instead.